circuitikz/circuitikz #19: Inconsistent direction of european and american voltages

gonike, 2015-11-05:
European and american voltages seem to behave inconsistently regarding the voltage direction. See [this github issue](http://tex.stackexchange.com/questions/37182/voltage-sign-convention-european-vs-american-in-circuitikz) (not asked by me). Voltage arrows across, e.g., resistors, as well as voltage source symbols are drawn in different directions using the two conventions.

sistlind, 2016-02-07:
You are right, the drawing directions for voltage and current are not really consistent. I will try to fix that with an option to preserve the old behavior.

sistlind, 2016-02-07:
I hope a fixed this issue, please have a look at the manual for the new counting directions.

Please reopen issue if something goes wrong

closed 2016-02-07 by sistlind

gonike, 2016-02-08:
Nice, thanks! :-)