circuitikz/circuitikz #24: Clean up documentation

der-stefan, 2016-03-18:
The documentation gets more and more confusing because of the sorting by oneports, twoports etc. (or "poles"). I think the user is more interested in the function of a device and if it behaves as a twoport (syntax ``(0,0) to[device] (2,0);``) or as a node (syntax ``(0,0) node[device] {};``). Futhermore, he wants to know how he addresses the ports (anchor names).

Any ideas?

der-stefan, 2016-04-15:
A first step is done with cb46e0a. More to follow.

Rmano, 2019-02-14:
I will start to try to reorganize it, putting together the element, their usage, and the anchors. I will start with the new features and then propagate it if accepted...

Rmano, 2019-03-21:
#169 should let us close this issue. There are still things to do, but I think it's quite a step in the right direction.

commit 7521146a (2019-03-22): Merge pull request #169 from Rmano/manual

Complete reorganization of the manual.
Fixes #24 
Fixes #28
closed 2019-03-22 by Rmano