circuitikz/circuitikz #25: Option "noconnectors"

der-stefan, 2016-03-21:
Some symbols have very long connectors (worst cases are the antenna symbols!). Normally the user wants to draw his own lines, so the connectors the ports/anchors should be either very short or non-existent. We could add an option "noconnectors" to remove all long lines. For default nothing should change in order to keep backward compatibility for users.

Rmano, 2019-03-21:
I think this is related basically to antennas so it is quite easier to define no connector equivalents...

Rmano, 2019-03-21:
I will commit these antennas after the big manual merge...

[image]

I hope they will suffice.

Rmano, 2019-03-22:
Closed by #170

closed 2019-03-22 by Rmano