circuitikz/circuitikz #26: Add possibility to place Bipoles also as nodes

der-stefan, 2016-03-21:
Right now you cannot use bipoles also as nodes. Why not? Perhaps you want to be able to place an inductor with the syntax: ``\draw (0,0) node[L] {l}`` and use the anchors to connect it on your own.
The alignment could be managed by using the option ``anchor=...``, just like with the standard nodes.

osfe, 2016-07-06:
Dear all,
I totally agree with der-stefan. That would be a great feature to be able to instance bipoles as node
\node[shape=R,length=2cm] (r1) at (0,0) {};

I think that would make things easier, for example to place components in matrices...

Olivier

sistlind, 2016-07-06:
This should already be possible, e.g:
```
\begin{circuitikz}
\draw (0,0) node[resistorshape]{}
++(2,0) node[genericshape]{}
++(2,0) node[vcuteinductorshape]{}
++(2,0) node[vsourcesinshape]{};
\end{circuitikz}
```
Have a look at pgfcircbipoles.tex for the internal shapename and append "shape".

Best regards,
Stefan L.

closed 2017-07-08 by sistlind