circuitikz/circuitikz #32: American Voltage and Current Sources Can't Reverse Polarity

karlberggren, 2016-05-02:
I believe there is a bug wherein the american voltage sources cannot have polarity reversed using the v^>= and v^<= labeling operations.  A detail example is shown here:

https://www.overleaf.com/read/zsqbzpbfhhbq

I apologize if this was not the correct place to raise this issue.

sistlind, 2016-05-02:
Hi,
this should be already fixed in the latest release version 0.5 at commit 9309701, reported already as #19
At least at my local installation the output looks good:
[image]

Perhaps this version is not yet included in overleaf? It was released last week.
You can check the version with the command \pgfcircversion
Just reopen this ticket, if the issue exists still at version 0.5

karlberggren, 2016-05-02:
You were correct.  Overleaf is running version 0.4.  I should have checked
that before reporting.  Sorry for wasting your time, and thank you for the
helpful response (and for fixing this issue!).

Rmano, 2021-09-22:
Reporting the code here:

```latex
\documentclass[]{article}
\usepackage{circuitikz} 
\begin{document}

I believe I may have found a small bug in circuitikz.  You may 
already be aware of this.

When drawing American voltage sources, the polarity of the source does not switch back and forth with the <= and >= label operator.  An example of the situation is shown below, with the first set of symbols being european (where the expected behavior occurs) and the 2nd set being the american (where I would have expected the first two to have the - terminal on top instead of on the bottom).

\pgfcircversion

\begin{circuitikz} 
\draw
    (0,0) to [voltage source, v^<=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}
\draw
    (0,0) to [voltage source, v_<=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}
\draw
    (0,0) to [voltage source, v^>=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}
\pgfcircversion;
\draw
    (0,0) to [voltage source, v_>=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}

\begin{circuitikz}[american]
\draw
    (0,0) to [voltage source, v^<=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}[american]
\draw
    (0,0) to [voltage source, v_<=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}[american]
\draw
    (0,0) to [voltage source, v^>=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}[american]
\draw
    (0,0) to [voltage source, v_>=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}

the current sources seem to have the same issue

\begin{circuitikz}
\draw
    (0,0) to [current source, i^<=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}
\draw
    (0,0) to [current source, i_<=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}
\draw
    (0,0) to [current source, i^>=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}
\draw
    (0,0) to [current source, i_>=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}

\begin{circuitikz}[american]
\draw
    (0,0) to [current source, i^<=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}[american]
\draw
    (0,0) to [current source, i_<=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}[american]
\draw
    (0,0) to [current source, i^>=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}
\quad
\begin{circuitikz}[american]
\draw
    (0,0) to [current source, i_>=$\underline{U}_{in}$]
    (0,2)
;
\end{circuitikz}

\end{document}
```

[image]

Rmano, 2021-09-22:
Notice that the policy of `circuitikz` is not to make incompatible changes with older versions. So if you give no options to the loading of `circuitikz`, you have the old (that somebody thought buggy, somebody right) behavior.

If you apply `\usepackage[RPvoltages]{circuitikz}` the result will be different (but maybe not what you expect): the `v` label will *not* change the symbol orientation by design, only the *external* voltage label/sign. You need to use `invert` for the symbol internal signs. 

I know this is a bit of controversial thing, but it has been like that forever, and changing it will invalidate older circuit --- too big a change in my opinion.

ArchangeGabriel, 2021-09-22:
@Rmano: So it used to work at some point, then not in 0.4, then again from 0.5 to 0.8.2. So technically it has been changing several times already. Also I find id a bit sad to have four `v_>=`, `v_<=`, `v^>=` and `v^<=` commands for only two outputs…

But you’re right, I should be using RPvoltages, I actually use it in my code but when working on an MWE for tex.se I did not include it, could not find a way to have the voltage source in the correct direction and stumbled upon this issue.

Rmano, 2021-09-22:
@ArchangeGabriel yes, you right, it's a bit of a mess; you can see here: https://tex.stackexchange.com/questions/612425/change-circuitikz-vsource-current-arrow-direction#comment1531688_612437 (and in the following comments) what I think about it now. 

The four commands will have different outputs in non-generator devices, that you can force if you want (but I understand that in this case it has no sense).

ArchangeGabriel, 2021-09-22:
Well given the number of things that seems broken because of needing to keep retro-compatibility, I’d be in favour of a huge breaking release at this point, maybe using something like `mhchem`’s `version` key, or just a `modern` vs `legacy` (default) behaviour. But you’re the one in charge and doing actual development, so my opinion does not matter much. ;)

Rmano, 2021-09-22:
Well, that could be an option --- I just need to track what the changes should be. I really do not know much more apart from the eternal fight for voltage/current directions, and the detail if `v` should change the symbol direction... and in this case, I am not sure. Because now it's easy: symbols are never inverted by `v<`. To change this, I should add a flag to each symbol to say if it changes or not and a form to override. Consider for example: 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{tikzpicture}[european voltages]
    \draw (0,0) to[american voltage source, v=$v_1$] ++(0,2);
    \node at (1,1) {$=$};
    \draw (3,0) to[american voltage source, v<=$-v_1$] ++(0,2);
\end{tikzpicture}
\end{document}
```

[image]

If `v` changes the direction of the symbol, the thing will become complex. We should add a non-invert option for the second one...

Rmano, 2021-10-05:
Closing, feel free to reopen if you want to comment!

closed 2021-10-05 by Rmano