circuitikz/circuitikz #35: add alternate (conventional?) zener symbol

jason-s, 2016-05-05:
The zener symbol in circuitikz is barely distinguishable from a "regular" diode:

[image]

This is an odd symbol (not sure where it came from); the majority of zener diode manufacturers use the "zigzag" symbol instead:

http://www.onsemi.com/pub_link/Collateral/BZX84C2V4LT1-D.PDF
[image]

http://www.diodes.com/_files/datasheets/ds18001.pdf
[image]

https://www.fairchildsemi.com/datasheets/BZ/BZX84C10.pdf
[image]

Philips/NXP and Vishay use a variant that is more similar to the one in circuitikz, but it's different, the little tag on the end of the diode line points down, not up.

http://www.nxp.com/documents/data_sheet/BZX84_SERIES.pdf
[image]

https://www.vishay.com/docs/85763/bzx84v.pdf
[image]

gpitel, 2016-05-05:
For symbols I recommend following IEEE Electrical Symbol standards.

[ansii_graphic_symbols_for_electrical_and_electronics_daigrams_1993.pdf](https://github.com/mredaelli/circuitikz/files/250906/ansii_graphic_symbols_for_electrical_and_electronics_daigrams_1993.pdf)

jason-s, 2016-05-05:
good idea, but zener diodes and IGBTs aren't in that document.

jason-s, 2016-05-05:
Here's what I'm using as a workaround; you guys are welcome to integrate it into circuitikz as a variant. (I'm not exactly sure if I got it right the way you prefer, and I'm not sure how to add it to the circuitikz source.)

[image]

```
\documentclass[border=6mm]{standalone}
\usepackage{circuitikz}
\usepackage{tikz}

\makeatletter
\pgfcircdeclarebipole{}{\ctikzvalof{bipoles/diode/height}}{fullzzdiode}{\ctikzvalof{bipoles/diode/height}}{\ctikzvalof{bipoles/diode/width}}{

	\pgfsetlinewidth{\pgfkeysvalueof{/tikz/circuitikz/bipoles/thickness}\pgfstartlinewidth}

	\pgfscope
		\pgftransformxshift{\pgf@circ@res@left}
		\pgfpathmoveto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@left}{0pt}}
		\pgfpathlineto{\pgfpoint{0pt}{\pgf@circ@res@up}}
		\pgfpathlineto{\pgfpoint{0pt}{\pgf@circ@res@down}}
		\pgfpathlineto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@left}{0pt}}
		\pgfusepath{draw,fill}
		\pgfpathmoveto{\pgfpoint{\pgf@circ@res@right-1.8\pgf@circ@res@left}{\pgf@circ@res@down-0.5\pgf@circ@res@up}}
		\pgfpathlineto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@left}{\pgf@circ@res@down}}
		\pgfpathlineto{\pgfpoint{\pgf@circ@res@right-\pgf@circ@res@left}{\pgf@circ@res@up}}
		\pgfpathlineto{\pgfpoint{\pgf@circ@res@right-0.2\pgf@circ@res@left}{\pgf@circ@res@up-0.5\pgf@circ@res@down}}
		\pgfusepath{draw}
	\endpgfscope
}
\def\pgf@circ@fullzzdiode@path#1{\pgf@circ@bipole@path{fullzzdiode}{#1}}
\tikzset{full Zzener diode/.style = {\circuitikzbasekey, /tikz/to path=\pgf@circ@fullzzdiode@path}}
\tikzset{zzD*/.style = {full Zzener diode}}
\makeatother

\begin{document}
\begin{tikzpicture}[american, font=\sffamily]  
\ctikzset{bipoles/resistor/height=0.1}
\ctikzset{bipoles/resistor/width=0.3333}

\draw 
(0,0) to [/tikz/circuitikz/bipoles/length=0.6cm, zzD*] ++(0,1) to [R] ++(0,1);
;
\end{tikzpicture}
\end{document}
```

sistlind, 2016-05-05:
Hi,
we plan to rework a lot of symbols to comply with the european standard [EN-60617](http://pcad-libs.embedders.org/rules/ref_617.pdf) . Therefore, we will change at least the direction of this zener-tip.

Your version looks more "american", but it is easier distinguishable from the  schottky-diode,therefore,  i will try to implement also this one :+1:

commit 4c4d17d4 (2016-05-10): corrected shape of zener diode, added new alternative zigzag zener diode and improved shape of schottkydio(now better distinguishable from Zener due to longer lines); closes #35
closed 2016-05-10 by sistlind