circuitikz/circuitikz #41: Request: Option to place dots on transformer

mlda065, 2016-06-20:
Currently transformers look like this.

[image]

I would like to be able to add dots to indicate polarity. Like this.

[image]
[image]

gpitel, 2016-06-22:
Matt, 
I have the code for what you looking for below.  I used node[circ] -- but I do agree it would be nice to to have those dots built into the library.
http://onesolver.com/#!service=system_saved_load&id=16&tab=3

bastien-roucaries, 2018-02-27:
Indentically for inductance, it vill be nice to have dot

Rmano, 2018-11-25:
Would be nice, yes, but there are a couple of delicate things to take into account:
1. the color of the dot must follow the `color` option (not automatic, given the fact that this is a fill and not a stroke) 
2. it's important to choose a reasonable way to specify the dot position

sistlind, 2018-11-27:
In my opinio its the same problem as #80 . Just draw it by hand. This is much more flexible. Nevertheless, a good idea might be to declare 4 anchors at correct positions to simplify the placement.

Rmano, 2019-02-16:
Hmmm.... this seems a bug: 

[image]

I will fix it on the way along, hoping it will not break anything.

Rmano, 2019-02-16:
This is my proposal for the anchors:

[image]

They will scale nicely with the size of the elements... 

[image]

I'll prepare a PR with it, as soon as I add it to the manual.

commit e627c58e (2019-02-16): Merge pull request #153 from Rmano/transfdot

Add anchors to easily add dots to transformers; fixes #41
closed 2019-02-16 by Rmano