circuitikz/circuitikz #43: Request: invert bipoles option

Piolie, 2016-07-04:
When drawing non-symmetrical bipoles, it would be useful to able to invert the direction of the component. For example, when drawing a diode:
`(0,0) to[D] (2,0)`
it would be nice to have an option like `invert` or `reverse`, to make it behave as if
`(2,0) to[D] (0,0)`
was typed. Thus:
`(0,0) to[D, invert/reverse] (2,0)` would produce the same as `(2,0) to[D] (0,0)`
This option could also be extended to transistor (and other tripole) paths.
I know one could simply write the latter, but sometimes it can be tricky to change the way the circuit has been drawn to accomodate for this.

sistlind, 2016-07-04:
I am missing this feature,too.
I wiill try to implement it soon :)

sistlind, 2016-07-06:
It was already included in the code, but nowhere documented and with incorrect voltage and label positions, therefore i had to rewrite the most.
Now, the option is included in the latest git version 6095c81, please take the git version  from here and test it: http://circuitikz.github.io/circuitikz/

closed 2016-07-06 by sistlind