circuitikz/circuitikz #46: Error of the key '/tikz/piezoelectric' or the key '/tikz/PZ' in Ubuntu

khaaba, 2016-07-22:
I am using TeXworks for compiling LaTeX documents in Windows 10, MacOS and Ubuntu. So far, I have no problem with circuitikz codes in Windows and MacOS systems, however, I recently faced a problem while LaTeX in Ubuntu using PZ or piezoelectric component which gives an error:
`! Package pgfkeys Error: I do not know the key '/tikz/piezoelectric' and I am g
oing to ignore it. Perhaps you misspelled it.

See the pgfkeys package documentation for explanation.`

This relates to a thread @ http://tex.stackexchange.com/a/316412/18678.

sistlind, 2016-07-22:
Which circuitikz do you use? If you are using the texlive-packages from ubuntu, then they will be quite old.
Test your version with the command \pgfcircversion, and if it too old, install Texlive manually using the web installer or use the circuitikz-git version.
Grab the git-version here: https://circuitikz.github.io/circuitikz/circuitikzgit.sty, put it in your project directory and include it with \usepackage{circuitikzgit}

khaaba, 2016-07-22:
I used 'circuitikz' only which works comfortably on my MikTeX (on Windows 10) and MacTeX (on El Capitan). But the code is failed to work on Ubuntu. 
I doubted the update part earlier too. So I used `sudo apt-get install texlive-full` to update the same but the last sentence says `0 upgraded, 0 newly installed, 0 to remove and 17 not upgraded.` I don't know which could not upgraded. 
I even downloaded 'circuitikz.sty' from CTAN. Still didn't work. But your suggestion works with 'PZ' key however, I face another problem with other circuit elements: Tnpn, R etc. like for instance `! Undefined control sequence.
\pgf@sh@bg@npn -> \pgftransformationadjustments `

My MWE is given below:

`\documentclass[border=10pt]{standalone}

\usepackage{circuitikz}
%\usepackage{circuitikzgit}

\begin{document}

\begin{circuitikz}
    \draw (2,0)
     to [Tnpn,n=npn] (2,1.4)
        (npn.E) node[right=3mm, above=5mm]{$Q$}; % Label the transistor
    \draw (2,1) to (2,3) node[circ]{} 
%    to [PZ, l=$X_1$] (0,3)
        to [R,l_=$1\Omega$] (0,0.7) node[circ]{}
        to (1.2,0.7);
\end{circuitikz}

\end{document}`

sistlind, 2016-07-22:
As i  assumed, you are using the ubuntu packages, which are **really** old.
The problem is, that the ubuntupackages do not deliver the tlmgr-tool, therefore, there is no possibility to upgrade your texlive installation.

The Tnpn-error is due to an outdated version of tikz/pgf in combination with the circuitikz-git-version.
Please install Texlive2016 with the netinstall from here: https://www.tug.org/texlive/ and everything should work fine.

I know it is annoying, and there seems to be no ppa for the most recent texlive...

khaaba, 2016-07-25:
Thanks I got it!

closed 2016-07-25 by khaaba