circuitikz/circuitikz #49: American voltage source has backwards polarity

davidparent, 2016-08-17:
Environment: This is pdfTeX, Version 3.14159265-2.6-1.40.17 (MiKTeX 2.9 64-bit)
Windows 7, 64 bit
Circuitikz: Should be latest as I set Miktex to install libraries on the fly, and then I just compile the manual.

Even though the manual when complied, correctly placed the +, and - symbols for every instance an american voltage is used, for the exams I have attached the +, and - signs are reversed.

While I can program them to be reversed, I am using lcapy which generates the figures. The notebooks I want to make for my students will be a mess to teach with if I have to back end process the tex file lcapy via i-python creates.  (lcapy will actually simulate the circuit drawn so voltage drops matter.)

Overleaf used the same code and the polarity is plotted correctly.

[induction-machine.pdf](https://github.com/circuitikz/circuitikz/files/423308/induction-machine.pdf)
[parent1.pdf](https://github.com/circuitikz/circuitikz/files/423309/parent1.pdf)
[induction-machine.docx](https://github.com/circuitikz/circuitikz/files/423312/induction-machine.docx)
[parent1.docx](https://github.com/circuitikz/circuitikz/files/423313/parent1.docx)

sistlind, 2016-08-17:
I am not shure, if i understand your problem correctly and i don't know lcapy or overleaf.
Nevertheless, there were some bugs at the placement of the voltagelabel in versions prior to v0.5(?).
Unfortunately, fxixing these bugs leads sometimes to an inversion of the lables.
But there exists a packet-option "oldvoltagedirection" which forces using the old erronous voltage counting direction.
Please try this option for your old examples and use for new projects the  new drawing style, because it is easier to understand (drawing direction=positive current direction) and the result will be equal between european and american style.

davidparent, 2016-08-17:
Thanks, oldvoltagedirection did fix it, but now I would have to have the lcapy author change his code to use that or use the new style which might be hard.  I think the author of lcapy is using an old version of circuitikz. Overleaf is an online tex viewer I was using to see if it was a windows problem.  (I just saw your edit.)

davidparent, 2016-08-17:
Thanks, I think this issue should be closed because I know what to do, even if I do not want or can not do it.

closed 2016-08-17 by davidparent