circuitikz/circuitikz #53: Drawing inductors (cute or american) stopped working

jnareb, 2016-09-13:
Drawing inductors (whether with 'cuteinductors' or 'americaninductors' option) works on one machine (with older circuitikz), but does not work on other machine (with most current circuitikz).

```
! Undefined control sequence.
\pgf@sh@bg@americaninductorshape ...onadjustments
\advance \pgf@circ@res@ste...
l.470 \draw (3,1) to[L,l^=$L$] (3,-1)
```

```
! Undefined control sequence.
\pgf@sh@bg@cuteinductorshape ...mationadjustments
\par \pgfmathsetlength {\p...
l.470 \draw (3,1) to[L,l^=$L$] (3,-1)
```

PGF/TikZ from 2014-02-28

sistlind, 2016-09-13:
Please update your complete installation, your pgf/tikz version is too old.  This can be done using the miktex update tool. It could be, that you have to run the update several times due to your old installation.

sistlind, 2017-02-05:
Problem solved? If not, just reopen the issue

closed 2017-02-05 by sistlind