circuitikz/circuitikz #54: Request: Microstrip Transmission Lines

tcpluess, 2016-10-05:
The symbol for a coaxial transmission line is already available. However, there is also a wide usage of microstrip transmission lines and stubs. Two new symbols could be added: a bipole for the transmission lines. It has the same shape as an european resistor, but twice as long. Further, a microstrip stub is also needed, which has the same symbol as before, but with only one connection.

Is that possible?

Rmano, 2019-02-25:
Hi @tcpluess  (I know it's a long time...), your symbol is something like this:
[image]
...for which I can't see how the description fits:

> Two new symbols could be added: a bipole for the transmission lines. It has the same shape as an european resistor, but twice as long. Further, a microstrip stub is also needed, which has the same symbol as before, but with only one connection.

...given that an european resistor is just a rectangle. Could you please add an example symbol and how it is connected?

tcpluess, 2019-03-27:
sorry for my late reply as well. See the attached printscreen from a microwave CAD package. It shows an example of how to use the transmission lines symbols as well as the radial stub. In my original posting it appears as if I had posted some wrong code, don't know how that happened :-)

[image]

The rectangles and radial stubs don't represent actual circuit elements but transmission lines having a specified width and height. See another example on a PCB here:

https://en.wikipedia.org/wiki/Stub_(electronics)#/media/File:Microstrip_Low_Pass_Bowtie_Stub_Filter.jpg

This kind of symbol is required to draw schematics of commensurate circuits.
https://en.wikipedia.org/wiki/Commensurate_line_circuit

Rmano, 2019-03-28:
Ah, ok. 
These two components are not difficult to create, I'll see if I can add them.

Rmano, 2019-04-02:
Hi, I just realized that we have these two components: 

`tline`, `TL`: 
[image]

`tlinestub`: 
[image]

(Forget the colors, just my fiddling). Are they just another style?

sistlind, 2019-04-02:
one is a stub (end of tline), therefore, they are not te same

Rmano, 2019-04-02:
Yes; but the OP is asking for "simply square" elements, I think --- a transmission line which is simply a rectangle, a linear stub, and a radial stub. We have two of them but with a different shape; I am not into the RF electronics so I do not know if it's just another style or if the "cylindrical" ones we have are correct or whatever.

Rmano, 2019-04-12:
@tcpluess, what do you think of:
[image]

Rmano, 2019-04-12:
I will merge the change (which will close the issue). @tcpluess, feel free to comment here or reopen the issue if you need to.

PR #181 (2019-04-12): Added microstrip elements
Change suggested by @tcpluess at https://github.com/circuitikz/circuitikz/issues/54

Fixes #54
commit a9f8accd (2019-04-12): Merge pull request #181 from Rmano/rfthings

Added microstrip elements.
Fixes #54
closed 2019-04-12 by Rmano