circuitikz/circuitikz #56: Draw straight arrows for voltage instead of incurved

thejoker2, 2016-12-04:
In France (and I guess most of Europe) voltage arrows next to the dipole are straight, not incurved.

Would it be possible to replace incurved arrow by straight arrows?

Up to now, everybody has to draw arrows separately next to each dipole, which is quite time-consuming, and it removes the interest of the package.

Thanks a lot for all of us!

thejoker2, 2016-12-04:
Sorry, I've been just told that the "straightvoltages" option exists.
`\usepackage[european,straightvoltages]{circuitikz}`

But unfortunately the result is not as good as expected, by far:

[image]
[image]

sistlind, 2017-02-05:
Hi,
the straighvoltages are only in alpha-testing, because it is very complicated to auto-position the label and the arrow. It should look better if you use a much larger schematic. Nevertheless, there are a few bugs left(e.g. u_7). Alternatively, just name the node and create the voltage arrows by hand afterwards(This is the way i draw my schematics).

Btw: This seems to be only the correct way in france. Accourding DIN EN 60375:2004 and  IEC 60050 the implemented directions are correct (see https://de.wikipedia.org/wiki/Z%C3%A4hlpfeil (sorry, found it only in german))

Best regards,
Stefan

closed 2017-03-08 by sistlind