circuitikz/circuitikz #57: Drawing multiple resistors in one go stretches coordinates

ZerataX, 2017-01-20:
```tex
\draw (14,0) node[below] {$a$}
[vR,l=$R_2$] to ++ (0,6);
\draw (15.5,0) node[below] {$b$}
[R,l=$R_i$] to ++ (0,3);
\draw (15.5,3) to [L,l=$L$]++ (0,3);
\draw (17,0) node[below] {$c$}
[R,l=$R_i$] to ++ (0,2);
\draw (17,2) to  [L,l=$L$]++ (0,2);
\draw (17,4) to  [C,l=$C$]++ (0,2);
```

produces
[image]
while 
```tex
\draw (14,0) node[below] {$a$}
[vR,l=$R_2$] to ++ (0,6);
\draw (15.5,0) node[below] {$b$}
[R,l=$R_i$] to ++ (0,3) to  [L,l=$L$]++ (0,3);
\draw (17.0,0) node[below] {$c$}
[R,l=$R_i$] to ++ (0,2) to  [L,l=$L$]++ (0,2) to  [C,l=$C$]++ (0,2);
```
results in
[image]
with the last line not even being displayed on the document anymore.
the coordinates are completely off for whatever reason.
To use the latter I would have to adjust the coordinates:
14 -> 14
15 -> 15.61
16 -> 16.84
Furthermore it creates 3 times Ri pretty close to (0,0)
[image]

sistlind, 2017-02-05:
You did not really supple the full example, here looks everything fine without the dashed lines and the circs. Nevertheless, i think you are just using the wrong path-syntax. The word "to" should always be in front of the square brackets.

Please add a full example, or close this issue, if the syntax was the problem.

Best regards, 
Stefan

ZerataX, 2017-02-09:
Yes seems like my syntax was the problem, personally I would like it, if wrong syntax would result in an error.
Additionally this is the full example:
```tex
\documentclass[preview]{standalone}
\usepackage[a3paper]{geometry}
\usepackage{pgf}
\usepackage{pgfkeys}
\usepackage{circuitikz}
\usepackage{tikz}
\usepackage{siunitx}
\usetikzlibrary{fit}

\begin{document}
	\newcommand{\esourcetowattmeter}[1]{
		\draw (#1)node[label={[label distance=-1.25cm]40:\Large w}]{};
		\draw (#1.north) [short] to (#1.south);
		\draw (#1.west) [short] to (#1.east);

	}
	\begin{circuitikz}[esource,european]
		\draw (4,0) node[spdt,xscale=-1,yscale=1,anchor=in] (Soben) {};
		\draw (4,6) node[spdt,xscale=-1,yscale=1,anchor=in] (Sunten) {};
		
		
		\draw (0,-0.318) [short,o-] to (Soben.out 2) node {};
		\draw (0,0.318)	[short,o-] to (Sunten.out 2) node {};
		
		\draw (0,5.682) node[left] {$-$} 
		[short,o-] to (Soben.out 1) node {};
		\draw(0,6.318) node[left] {$+$}
		[short,o-] to (Sunten.out 1) node {};
		
		\draw (0,6) node[left] {$\SI{24}{\volt}$};
		\draw (0,0) node[left] {$\SI{24}{\volt}$};
		\draw (1.35,0) node[left] {$\sim$ $\SI{50}{\hertz}$};
		
		
		\draw (4,6) [short] to (6,6)
		to [short,vR, l_=$R_1$ ] (6,0);
		\draw (7.5,3.05) node[left] {$\SI{27}{\ohm}$};
		
		\draw (6.5,3.27) [short] to (6.5,6);
		
		
		\draw (6.5,6) to [short] (7,6) coordinate(left) to [esource,n=wattmeter]++(2,0) to [short,-*]++ (.25,0) [short,-*, l=3] to (10,6);;
		
		\draw (left)to [short,*-,l=1/2]++(0,0.75)-|(wattmeter.north);
		\draw (wattmeter.south) to [short,-*,l=5] (8,5) [short ]to (8,0);
		
		\esourcetowattmeter{wattmeter}
		
		
		\draw (10,6) [voltmeter] to (10,0);
		\draw (10,6) [ammeter,-o] to (13,6);
		
		
		\draw (13,6) [short,densely dashed] to [short,-*]++ (1,0)
		[short,densely dashed] to [short,-*]++ (1.5,0)
		[short,densely dashed] to [short,-*]++ (1.5,0);
		\draw (13,0) [short,densely dashed] to [short,-*]++ (1,0)
		[short,densely dashed] to [short,-*]++ (1.5,0)
		[short,densely dashed] to [short,-*]++ (1.5,0);
		
		\draw (4,0) [short,-*] to (6,0)
		to [short,-*] (8,0)
		to [short,-*] (10,0)
		to [short,-o] (13,0);
		
		\draw (14,0) node[below] {I}
		to [vR,l=$R_2$] (14,6);
		\draw (15.5,0) node[below] {II}
		to [R,l=$R_i$] ++ (0,3) to [L,l=$L$]++ (0,3);
		\draw (17,0) node[below] {III}
		to [R,l=$R_i$] ++ (0,2) to [L,l=$L$]++ (0,2) to [C,l=$C$]++ (0,2);

%		\draw (14,0) node[below] {I}
%		[vR,l=$R_2$] to ++ (0,6);
%		\draw (15.5,0) node[below] {II}
%		[R,l=$R_i$] to ++ (0,3);
%		\draw (15.5,3) to [L,l=$L$]++ (0,3);
%		\draw (17,0) node[below] {III}
%		[R,l=$R_i$] to ++ (0,2);
%		\draw (17,2) to  [L,l=$L$]++ (0,2);
%		\draw (17,4) to  [C,l=$C$]++ (0,2);
	\end{circuitikz}
\end{document}
```

closed 2017-02-09 by ZerataX