circuitikz/circuitikz #58: Connection corners, dipole

iuliuv, 2017-01-21:
When connecting dipoles at right angle, the connection is not nice:
`\draw (0,0) to[R] (0,2) to[R] (2,2);`
the same connection with wires looks much better:
`\draw (0,0) -- (0,2) -- (2,2);`
Am I missing something? Any easy way to avoid the issue? If nothing else maybe define a node that draws a filled square of linewidth size as a temporary solution? (similar to `node[circ]{}` )
thanks!

sistlind, 2017-02-12:
The commit breaks support of relative coordinates at paths--> old behavior reverted and so solution at all

sistlind, 2017-03-07:
Please try:
`\draw (0,0)to[R,-.]++(0,2)to[R]++(2,0);`
this can be combined with *,o,d as regular pole shape.

commit 3a1205d9 (2017-02-05): dirty hack to fix #58, hope to find a better solution
commit 352d2bf3 (2017-03-08): Fix #58: Line joins should be finally fixed
PR #418 (2020-07-01): Rewrite path formation
This basically removes the patches introduced for issue #58 and #76
    
Reverts these:
-  https://github.com/circuitikz/circuitikz/commit/6efc2ee86fb3b2b479b3039526fcc90d3f9201d2
-  https://github.com/circuitikz/circuitikz/commit/352d2bf34054d6450a0ffe92529876f4e2398ac6
    
Then, it carefully builds the anchors without moving the start of the path, and then join it with a leading `--`

Fixes #417 
Fixes #76 
Fixes #58
closed 2017-03-08 by sistlind