circuitikz/circuitikz #59: Request: Floating Current Arrows

srhathorn, 2017-02-14:
Many books and circuit diagrams use floating arrows to indicate current; e.g., http://imgur.com/a/f76Ow from CMOS Circuit Design by Baker p. 867. 

It seems this is a European Style vs American Style difference. I am able to make the arrows manually using tikz, but it would be nice to be able to use it in the same method as the on wire arrows.

sistlind, 2017-03-08:
How do you draw voltages? As a workaround, you can use a short in combination with the "straightvoltages" option. This is possible since the latest git-commit 248e372e3e0e02e0b239879a8ee2a1be86abac82

Best regards, Stefan

srhathorn, 2017-03-20:
I've had a functional workaround for a while. 
Voltages are done using the +/- sign already implemented for american voltages. 
Here's an example with voltage and current, from another textbook: 
[image]

abalmos, 2017-05-12:
@sistlind Can you offer any guidance on the best way to add this to circuitikz? We can try to contribute this feature.

sistlind, 2017-05-19:
Hi, 
at the moment, i have no time to implement this, but i will willingly include any pull request on this topic. I think it is not to complicated to implement this feature. Up to now, only a currarrow is placed at the line using this code:
https://github.com/circuitikz/circuitikz/blob/master/tex/pgfcirccurrent.tex#L195

You just have to implement an switch/option(Have a look at the option keys at circuitikz.sty how this can be accomplish) and the placement of the currarrow have to be replaced by a short line with an arrow(just a new symbol, like the currarrow) and an offet(yshift at https://github.com/circuitikz/circuitikz/blob/master/tex/pgfcirccurrent.tex#L195).

The main problem will be testing and optimizing the placement for all circumstances :)

If you have any further questions, feel free to ask!

Best regards, stefan

sistlind, 2017-05-28:
I have though on this, perhaps it is the best to call this "flow", as it can also be used as power flow, thermal flow, or something else.
It will then be called by the keys f=, or flow= .

What do you think about this solution?

sistlind, 2017-05-28:
I have implemented this feature, pleas test it. Documentation will follow, if everything works ;)

commit a50f31c3 (2017-05-28): New annotation type to show flows using 'f=...' like currents. It can be used for thermal, power or current flows (closes #59)
closed 2017-05-28 by sistlind