circuitikz/circuitikz #63: Unable to use > to create arrow head with "Basic resistive bipoles"

ErikThorsell, 2017-03-23:
I was trying to draw an arrow head pointing into a generic block, however this seems not to work.

**Steps to recreate**:
```
\begin{circuitikz}
    \draw
        (0,0) to[short, -] ++(0.3,0)
        to[generic,>] ++(2,0)
    ;
\end{circuitikz}
```
I would expect to see a generic box with an arrow head on its left, however there is simply a box. Changing `generic` to a component not in the `Basic resistive bipoles` yields the expected result. I see no notice of this in the documentation, hence I would call this an issue.

ErikThorsell, 2017-03-23:
The above is applicable to resistors as well. (I.e. arrow heads are not working.)

sistlind, 2017-03-24:
Where should the arrawhead apper? at the resistor or at the end of the path? Could you draw a sketch, how it should look like?

ErikThorsell, 2017-03-30:
Sorry for a late reply. I would like the arrow to be "touching" the box. If you for instance use a `node` and an `inputarrow`, you get an arrow pointing at the edge of the node.
```
\begin{circuitikz}
    \draw
        (5,0) node[adder] (m) {}
        (m.1) to[short] ++(-1,0)
        (m.4) to[short] ++(0,1)
        (m.1) node[inputarrow] {}
        (m.4) node[inputarrow, rotate=-90] {}
    ;
\end{circuitikz}
```
I am currently not able to provide a sketch and I have dropped this little project for the moment, but I'll get back when the need arises again!

sistlind, 2017-05-28:
If you want a touching arrow, you can try this approach using a name path-object:
```
\begin{tikzpicture}
	\draw (0,0)to [R,n=shapename]++(2,0);
	\draw (shapename.left) coordinate[inputarrow];
\end{tikzpicture} 
```
[image]

Please reopen, if this does not fits your need

closed 2017-05-28 by sistlind