circuitikz/circuitikz #66: american thermistor

sistlind, 2017-04-09:
See http://tex.stackexchange.com/questions/362754/circuitikz-american-thermistor-symbol

Thermistor symbol should use european/american resistorshape depending on packetoption

Rmano, 2018-11-25:
I have a pull request for a generic sensor at #124 : is that ok for you? It is not automatic (yet), but the shape is there...

mredaelli, 2018-11-25:
@Rmano Seems good. I hope to have time to checkout everything soon to test it and get up to speed.
Two things:
- For consistency's sake, might it be better to call it american thermistor? Or do we reserve that for the symbol with also the label as per SE picture? I'm not familiar...
- In general, I see the same commits in many of your PRs. Can we factor them out?

Rmano, 2018-11-26:
> * For consistency's sake, might it be better to call it american thermistor? Or do we reserve that for the symbol with also the label as per SE picture? I'm not familiar...

Here we normally use a lot the generic symbol, without labels, for a generic resistive sensor (could be a thermistor, could be whatever other things). Maybe the "thermistor" (with the label) could be added also.  

> * In general, I see the same commits in many of your PRs. Can we factor them out?

Yep, I messed up a bit with git (still learning). But it merges ok locally (I did it on my fork) with just a couple of minor conflict (not due to the commits in question, git seems smart enough to factor it out). My plan would be to merge locally and then push.

Rmano, 2018-11-28:
For the label/nolabel problem, I stolen the idea form @sistlind  about the dots on transformer, and what I did is adding a `label` anchor to `resistive sensor`:

```latex
\draw (2,0) to[sR, l=$R$, name=mySR] ++(3,0);
\node [font=\tiny, right] at(mySR.label) {-t\si{\degree}}; 
```
[image]

which will leave a lot of freedom. Now, the other thing that can be done is 
a) rename this to `american resistive sensor`
b) create an alias for `thermistor ntc` as `european resisitive sensor` (although it's not exact), and add the anchor `label` also to it 
c) create a choice (I am not sure how to do it, but I'll investigate) so that `resistive sensor` and `sR` switch automatically between european and american style

Opinions?

Rmano, 2018-11-28:
Done. See: 

```latex
\begin{circuitikz}[american]
    \draw (2,0) to[sR, l=$R$, name=mySR] ++(3,0);
    \node [font=\tiny, right] at(mySR.label) {-t\si{\degree}};
\end{circuitikz}

\begin{circuitikz}[european]
    \draw (2,0) to[sR, l=$R$, name=mySR] ++(3,0);
    \node [font=\tiny, right] at(mySR.label) {-t\si{\degree}};
\end{circuitikz}
```
[image]

I will update the manual for this and update the pull request.

Rmano, 2018-11-28:
...and by the way, I added `inductive sensor`s and `capacitive sensor`s, while at it. The inductive ones are changing with the european/american/cute style (although I can't see why anyone would use anything different from cute...)