circuitikz/circuitikz #68: PGFKeys error in parsing color

fuzzybear3965, 2017-04-19:
PGFPlots supports an alternative format for specifying colors that's not terribly popular but [is used by](https://github.com/JuliaPlots/Plots.jl/blob/master/src/backends/pgfplots.jl#L111) `Plots.jl` (which implicitly works with `PGFPlots.jl`, if desired). This format is documented in 4.7.5 of the [CTAN documentation of PGFPlots](http://ctan.mirror.rafal.ca/graphics/pgf/contrib/pgfplots/doc/pgfplots.pdf).

This specification of colors seems to "confuse" circuitikz. When used in conjunction with circuitikz (regardless of the inclusion of `[compatibility]` since `\usepackage{pgfplots}` may load `tikz` implicitly... I don't know) errors of the following form are generated: `test.tex:34: Package pgfkeys Error: I do not know the key '/tikz/1:red' and I am going to ignore it. Perhaps you misspelled it`. Now, I understand that this seems like a PGFPlots /pgfkeys problem since the the `pgfkeys` package is reporting the error. However, commenting out `\usepackage{circuitikz}` and all circuits results in a successful compilation. For example,

_test.tex_
```latex
\documentclass{article}
%\usepackage[compatibility]{circuitikz}
\usepackage{pgfplots}
\begin{document}

%\begin{tikzpicture}
   %\draw node[ground]{} (0,0)
   %to [*-] (0,1)
   %to [*R] (3,1)
   %to (3,0)
   %to node[ground]{} (3,0);
%\end{tikzpicture}

\begin{tikzpicture}[]
   \begin{axis}[
      {axis background/.style={fill={rgb,1:red,1.00000000;green,1.00000000;blue,1.00000000}}}
      ]\addplot+ [
      color = {rgb,1:red,0.88887350;green,0.43564919;blue,0.27812294},
      mark options = {
         color = {rgb,1:red,0.00000000;green,0.00000000;blue,0.00000000}, draw opacity = 1.0,
         fill = {rgb,1:red,0.00000000;green,0.60560316;blue,0.97868012}, fill opacity = 1.0,
         },
      mark size = 2.0]coordinates {
         (0.0, 0.0)
         (1.0, 1.0)
         (2.0, 0.0)
      };
   \end{axis}
\end{tikzpicture}
\end{document}
```
compiles successfully. However, removing the comments and attempting to recompile results in 
```
Latexmk: applying rule 'pdflatex'...
This is pdfTeX, Version 3.14159265-2.6-1.40.18 (MiKTeX 2.9.6300 64-bit)
entering extended mode
(test.tex
LaTeX2e <2017-04-15>
Babel <3.9r> and hyphenation patterns for 75 language(s) loaded.
(C:\Miktex2.9\tex\latex\base\article.cls
Document Class: article 2014/09/29 v1.4h Standard LaTeX document class
(C:\Miktex2.9\tex\latex\base\size10.clo))
(C:\Miktex2.9\tex\latex\circuitikz\circuitikz.sty
(C:\Miktex2Latexmk: Log file says output to 'test.pdf'
Latexmk: Failure to make the files correctly
    ==> You will need to change a source file before I do another run <==
------------
Running 'C:\Users\JohnRinehart\Programs\vim_64\gvim.exe --servername GVIM --remote-expr vimtex#compiler#callback(0)'
------------
Collected error summary (may duplicate other messages):
  pdflatex: Command for 'pdflatex' gave return code 1
      Refer to 'test.log' for details
yer\pgfcore.sty
(C:\Miktex2.9\tex\latex\graphics\graphicx.sty
(C:\Miktex2.9\tex\latex\graphics\keyval.sty)
(C:\Miktex2.9\tex\latex\graphics\graphics.sty
(C:\Miktex2.9\tex\latex\graphics\trig.sty)
(C:\Miktex2.9\tex\latex\graphics-cfg\graphics.cfg)
(C:\Miktex2.9\tex\latex\graphics-def\pdftex.def
(C:\Miktex2.9\tex\generic\oberdiek\infwarerr.sty)
(C:\Miktex2.9\tex\generic\oberdiek\ltxcmds.sty))))
(C:\Miktex2.9\tex\latex\pgf\systemlayer\pgfsys.sty
(C:\Miktex2.9\tex\generic\pgf\systemlayer\pgfsys.code.tex
(C:\Miktex2.9\tex\generic\pgf\utilities\pgfkeys.code.tex
[... 220 lines cut ...]
Type  H <return>  for immediate help

test.tex:28: Package pgfkeys Error: I do not know the key '/tikz/0.00000000;blu
e' and I am going to ignore it. Perhaps you misspelled it.

See the pgfkeys package documentation for explanation.
Type  H <return>  for immediate help

test.tex:28: Package pgfkeys Error: I do not know the key '/tikz/0.00000000' an
d I am going to ignore it. Perhaps you misspelled it.

See the pgfkeys package documentation for explanation.
Type  H <return>  for immediate help
[1{C:/Users/JohnRinehart/AppData/Local/MiKTeX/2.9/pdftex/config/pdftex.map}]
(test.aux) )
(see the transcript file for additional information)<C:/Miktex2.9/fonts/type1/p
ublic/amsfonts/cm/cmmi10.pfb><C:/Miktex2.9/fonts/type1/public/amsfonts/cm/cmr10
.pfb>
Output written on test.pdf (1 page, 19493 bytes).
SyncTeX written on test.synctex.gz.
Transcript written on test.log.
=== TeX engine is 'pdfTeX'
Latexmk: Errors, so I did not complete making targets
```
I'm thinking that `circuitikz` redefines a pgfkey that then is used in order to parse the colors erroneously.

sistlind, 2017-04-19:
You are right, there are some missing curly brackets [(here)](https://github.com/circuitikz/circuitikz/blob/master/tex/pgfcirc.defines.tex#L839), will be fixed in a moment.

closed 2017-04-19 by sistlind

fuzzybear3965, 2017-04-19:
@sistlind, sorry this took me some time to get around to verifying. Yes! That worked! Thank you so much. I am using `circuitikzgit` right now. Are you opposed to tagging this to release a new version (so I can update MiKTeX and get the latest version)?

sistlind, 2017-04-20:
I will release a new version, if there are some more changes, because a ctan-release is quite complicated. For the while you can use the git-version, or (if you need it in several projects) correct the file within you local texmf-tree. Just search for pgfcirc.defines.tex and change it according to the commit.

commit 991dbef8 (2017-04-19): Fixes #68: pgfskeys error specifying mixed colors