circuitikz/circuitikz #70: european operational amplifier

claude-chevassu, 2017-04-30:
Hello,

I would like to have a symbol of operational amplifier to the european standards like shown on the drawing as an attachment.
I looked closely at "pgfcirctripoles.sty" in order to modify the symbol of the operational amplifier already present, but unfortunately I did not understand anything at all.
[image]
Could you help me please ? Thank you in advance.
Claude Chevassu

sistlind, 2017-05-01:
Yes, i can implement this type of opamp, but which is the correct symbol? I did not found your symbol, but those:
[image]
[image]
[image]

I don't have access to en/din standards, therefore i need some help here :)

claude-chevassu, 2017-05-01:
Thanks you for your quick response. You are right, the correct symbol is
not exactly the one I had attached, but rather the one that appears in the
attachment of this message, corresponding to the first you sent (a).

I thank you in advance.
Best regards,
Claude Chevassu

sistlind, 2017-05-01:
Is the implented version in http://circuitikz.github.io/circuitikz/circuitikzmanualgit.pdf ok for you?

claude-chevassu, 2017-05-01:
> Is the implented version in http://circuitikz.github.io/circuitikz/
> circuitikzmanualgit.pdf ok for you?

Very fine, it's ok for me. Is it already availble in circuitikz ?
I thank you very much.

I have a question. On this site (
https://tex.stackexchange.com/questions/19204/circuitikz-create-new-component)
Massimo Redaelli advise to impliment or modify the component in "
pgfcirctripoles.sty".
But in my computer, I can't find "pgfcirctripoles.sty" there is no such
file neither in circuitikz nor elsewhere, I've found in all Texlive, I can
only find "circuitikz.sty". But when I download "circuitikz" from CTAN,
there is "pgfcirctripoles.sty" in a directory named "circuitikz" in
addition to circuitikz.sty. But nevertheless texmaker works fine with
circuitiz on my computer and produce nice diagrams ! Very strange ! How
explain that ?

Best regards,
Claude Chevassu

closed 2017-05-01 by sistlind

sistlind, 2017-05-01:
Hi, ok i will close this and prepare a new release. It will be available at ctan in a few days.

You should find all files, which exist in this github-repository also at your local installation, although it is quite hard to find them due to the crude filesystem structure of tex. At a texlive-unix installation, the files should be here: /usr/local/texlive/2016/texmf-dist/tex/latex/circuitikz

Nevertheless, if you want to add something new, you can also put in in a normal file in you local tex-project folder and include it with input.

Best regards, Stefan