circuitikz/circuitikz #71: Shapes should be filled by color

sistlind, 2017-05-03:
All closed shapes should be fillable by a defined color.
https://tex.stackexchange.com/questions/361175/coloured-symbols-from-circuitikz

Rmano, 2018-12-04:
A possible solution could be to add keys `fill` and `fill opacity` to every closed shape, and then define a `\pgr@circ@fillordraw` that would read the key and apply it; then use this instead of the `\pgfusepath`. Quite a lot of work (and prone to error). 

By the way, is there a way if I am writing code for a shape to say "get the key for this shape named `fill`" without explicitly using the full path? I mean, instead of using `\pgfkeysvalueof{/tikz/circuitikz/bipoles/whatever/fillcolor}` is it possible to say something like `\pgfkeysvalueofthisshape{fillcolor}`? Ideally, a function searching upwards[1] would be great... If yes, the above strategy could be simpler...

[1] meaning that if `tikz/circuitikz/bipoles/whatever/fillcolor` does not exist, it will search `tikz/circuitikz/bipoles/fillcolor` and so on...

Rmano, 2019-02-28:
@mredaelli @sistlind This: https://tex.stackexchange.com/a/110083/38080 is a very interesting post. 

Should we change all the `\backgroundpath` to `\beforebackgroundpath`? We always do the stroke and fill ourselves...

Rmano, 2019-03-01:
So look, as a small experiment: changing in the definition of `buffer` shape the line that says `\pgfusepath{draw}` with:
```
                \pgfpathclose{} % that was a bug, must close the path
                \pgfextra{\typeout{FILLCOLOR:\tikz@fillcolor}}
                \ifx\tikz@fillcolor\pgfutil@empty
		    \pgfusepath{draw}
                \else
                    \pgfsetfillcolor{\tikz@fillcolor}
                    \pgfusepath{draw, fill}
                \fi
```
compiling: 

```
\begin{circuitikz}[
    ]
    \draw (0,3) node[buffer]{A};
    \fill[cyan] (-1,2) rectangle (2,0);
    \draw (0,1) node[buffer, fill=white]{A};
    \draw (1.2,1) node[buffer, fill=white, fill opacity=0.5, text opacity=1]{A};
\end{circuitikz}
```
gives:

[image]

which seems pretty it, ¿no? 

Still puzzled by the difference between `\backgroundpath` and `\behindbackgroundpath`...

Rmano, 2019-03-01:
So now my question is... which are the shapes that must be fillable? All the ones with a kind of closed path in them? Obviously amplifiers, chips and logic gates, but also empty diodes? RF components?

sistlind, 2019-03-01:
I vote for all closed shapes, e.g. at least  also resistors and diodes,
sources

Rmano, 2019-03-01:
Yes, that's reasonable. I think that the "black" component must stay black --- the are supposed to be filled with the draw color. Is that ok?

Rmano, 2019-03-01:
It's quite a long list, but the change is not difficult. I defined 
```
\def\pgf@circ@draworfill{%
    \ifx\tikz@fillcolor\pgfutil@empty
        \pgfusepath{draw}
    \else
        \pgfsetfillcolor{\tikz@fillcolor}
        \pgfusepath{draw, fill}
    \fi
}
```

and it's just a matter of checking where to put it instead of a `\pgfusepath{draw}` (and optionally correctly close the path, which is also good for the joining lines shape).

Rmano, 2019-03-01:
I mean: suppose you have: 

``` 
\draw [fill=white, fill opacity=0.5,] (4,0) to [D] ++(0,2) (5,0) to[D*] ++(0,2)
                                         (6,0) to[D*] ++(0,2) (7,0) to[D] ++(0,2) ;
```
if I do the changes at `emptydiode` I get: 

[image]

and as you can notice, the fulldiode is "doubly" transparent... and looking strange. I can fix this by adding a 
`\begin{pgftransparencygroup}`  `\end{pgftransparencygroup}` around the `\pgfusepath{draw,fill}` in the full diode, but now I am not sure if it's the right thing to do... 

[image]

Rmano, 2019-03-03:
I think I have a list of all fillable elements, and a file to test them. If someone can double check... (hint)... 

```
\documentclass[a4paper]{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage{a4wide}
\usepackage[RPvoltages, siunitx]{circuitikz}
\def\fillableto{%
    ammeter,
    voltmeter,
    ohmmeter,
    lamp,
    bulb,
    generic,
    tgeneric,
    ageneric,
    memristor,
    european resistor,
    variable european resistor,
    european potentiometer,
    european resistive sensor,
    varistor,
    phR,
    thR,
    thRp,
    thRn,
    fuse,
    afuse,
    diode,
    sDo,
    zDo,
    zzDo,
    tDo,
    pDo,
    leDo,
    VCo,
    biDo,
    D-,
    sD-,
    zD-,
    zzD-,
    tD-,
    pD-,
    leD-,
    VC-,
    Tr,
    Tro,
    Ty,
    Tyo,
    Ty-,
    european gas filled surge arrester,
    american gas filled surge arrester,
    ecapacitor,
    PZ,
    TL,
    european voltage source,
    vsourceC,
    american voltage source,
    european current source,
    isourceC,
    american current source,
    sV,
    sI,
    sqV,
    tV,
    esource,
    pvsource,
    ioosource,
    voosource,
    dcvsource,
    mass,
    twoport,
    vco,
    bandpass,
    bandstop,
    highpass,
    lowpass,
    adc,
    dac,
    dsp,
    fft,
    amp,
    vamp,
    piattenuator,
    vpiattenuator,
    tattenuator,
    vtattenuator,
    phaseshifter,
    vphaseshifter,
    detector,
    european controlled voltage source,
    cceV,
    american controlled voltage source,
    european controlled current source,
    european controlled current source,
    cceI,
    american controlled current source,
    csV,
    csI%
}
\def\fillablenode{sground,
    tlinestub,
    magnetron,
    mixer,
    adder,
    oscillator,
    circulator,
    elmech,
    american and port,
    american or port,
    american nand port,
    american nor port,
    american xor port,
    american xnor port,
    american not port,
    european and port,
    european or port,
    european nand port,
    european nor port,
    european xor port,
    european xnor port,
    european not port,
    schmitt,
    invschmitt,
    op amp,
    en amp,
    fd op amp,
    gm amp,
    inst amp,
    fd inst amp,
    inst amp ra,
    plain amp,
    buffer,
    ocirc,
    dipchip,
    qfpchip%
}
\begin{document}
\textbf{TO ELEMENTS}

\medskip

\foreach \eto in \fillableto {
    \parbox{3cm}{\raggedright\texttt{\eto}:}
    \begin{circuitikz}[baseline]
        \draw (0,0) to[\eto] ++(2,0);
        \fill[cyan] (3,-0.5) rectangle (8,0.5);
        \draw[fill=white] (3,0) to[\eto] ++(2,0);
        \draw[fill=white, fill opacity=0.4] (6,0) to[\eto] ++(2,0);
    \end{circuitikz}
    \par
}

\textbf{NODE ELEMENTS}

\medskip

\foreach \eno in \fillablenode {
    \parbox{3cm}{\raggedright\texttt{\eno}:}
    \begin{circuitikz}[baseline]
        \fill[cyan] (-1,-0.8) rectangle (8,0.8);
        \draw (1,0) node[\eno]{};
        \draw (4,0) node[\eno, fill=white]{};
        \draw (6,0) node[\eno, fill=white, fill opacity=0.4]{};
    \end{circuitikz}
    \par
}
\end{document}
```
PDF file: 
[fillable.pdf](https://github.com/circuitikz/circuitikz/files/2923528/fillable.pdf)

Rmano, 2019-03-03:
...`elmech` is behaving in strange ways...

Rmano, 2019-03-04:
Exploring further, I am seeing that 
- the fact that we stroke the path break the path-filling feature of `tikz`, fortunately, otherwise we will have a problem with `to`-style element. I hope it will have no bad effects somewhere else; 
- `ocirc` will be a problem; it is explicitely filled  with white, so that the `o-o` specifiers works, but this means it can't be transparent. I can't find (for now) where the `o-o` options are managed; the ideal thing would be not rely on center anchor in connecting, but using the border anchors of ocirc and leaving the shape unfilled.... @mredaelli, do you think it's possible? 

Example to show what I mean (this is using my `\pgf@circ@draworfill` for diodes and ocirc:
```
    \fill[cyan] (-1,4) rectangle (7,8);
    \draw [fill = yellow, thick] (0,5) -- ++(0,2) -- ++(2,0);
    \draw [fill=yellow, ] (4,5) to [D] ++(0,2) to[D*] ++(2,0)
    to[D*] ++(0,-2)  to[D, fill=red] ++(-2,0) -- cycle;
    \draw (1,4.5) to[short, o-o] ++(1,0) to[short, -o] ++(1,0);
    \draw[fill=yellow] (1,5) to[short, o-o] ++(1,0) to[short, -o] ++(1,0);
    \draw (1,5.5) to[short, fill=red, o-o] ++(1,0) to[short, -o] ++(1,0);
```

[image]

commit 6796e20e (2019-04-08): Merge pull request #175 from Rmano/fill

Fillable shapes: fixes #71
closed 2019-04-08 by Rmano