circuitikz/circuitikz #72: feature proposal -- generic block for block diagram

osfe, 2017-05-05:
Hi there,
I couldn't find a place to put this suggestion so I ask for it here. 
What would you think of a generic block, that could be used in electrical block diagram like we can see in datasheets? Say you want a generic block that has 3 inputs and two outputs and a name: 
- inputs
  + input 1 is a clock so it gets the name clock on which I can add the usual little triangle for clocked inputs
  + input 2 is an active low enable signal on which I can add a little bubble that shows active low signal 
  + input 3 is another signal called sig
- outputs 
  + output 1 and 2 are output signals

the block size could be chosen like I'd do with a generic node, I/Os could be distributed all along the side or put together on the upper/center/lower part of the block with a chosen distance between them. Inputs are on the left side, outputs on the right side... Pin names would be inside the block in front of the pin access node.

Any time I use circuitikz, which is a lot, I always miss this feature.

I know I'm asking a lot, but on the other side, it's just a suggestion :)

thank you for your work.

Olivier

Rmano, 2018-11-25:
I normally use a generic `tikz` node for this feature (too many options, I think...).

Rmano, 2019-02-15:
We will have chips now to do something similar. Say: 

```
\begin{circuitikz}
    \ctikzset{multipoles/thickness=4}
    \ctikzset{multipoles/dipchip/width=2}
    \draw (0,0) node[dipchip,
    num pins=10, hide numbers,
    external pins width=0,
    ](C){Block};
    \draw (C.pin 1) -- ++(-2,0) to[R] ++(0,-3) node[ground]{};
    \node [right, font=\tiny] at (C.bpin 1) {RST};
    \node [right, font=\tiny] at (C.bpin 2) {IN1};
    \node [right, font=\tiny] at (C.bpin 4) {IN2};
    \node [left, font=\tiny] at (C.bpin 8) {OUT};
    \draw (C.bpin 8) to[short,-o] ++(0.5,0);
\end{circuitikz}
```
[image]

...I can think about an option for removing the "upright marker" if needed.

Rmano, 2019-02-16:
Done, now adding `no topmark` to the chip, you can have: 

[image]

commit de949124 (2019-02-16): Merge pull request #150 from Rmano/chips

Add multi-pin chips; fixes #13 and fixes #72.
closed 2019-02-16 by Rmano