circuitikz/circuitikz #73: straightvoltages option and voltage arrow distance from bipoles

ckuhlmann, 2017-05-06:
Hi,
I'm trying to move the voltage arrows away from the bipole, but I cannot get it to work properly with the straightvoltages option enabled:
```
\documentclass[tikz,margin={2pt 2pt 2pt 2pt}]{standalone}
\usepackage[straightvoltages]{circuitikz}

\ctikzset{
%	/tikz/circuitikz/bipoles/resistor/voltage/distance from node/.initial=5,
%	voltage/european label distance/.initial=100,
	voltage/european label distance=100,
	bipoles/resistor/voltage/european label distance/.initial=100,
}

\begin{document}%
	\begin{tikzpicture}%
  \draw (2.5, -3.5) to[resistor=5R, l=$R_1$, a=6k, i=5A, v=U1] (5.0,-3.5){};  
	\end{tikzpicture}
\end{document}

```
I'm sorry if this is a silly mistake on my end.

sistlind, 2017-05-06:
Hi,
try this:
```
\ctikzset{
  bipoles/generic/voltage/straight label distance/.initial=.8,
  }
\begin{tikzpicture}%
  \draw (2.5, -3.5) to[R=5R, l=$R_1$, a=6k, i=5A, v=U1] (5.0,-3.5){};  
\end{tikzpicture}```

(The resistor-shape is internally called generic)
The straightvoltages are a bit beta, especially in combination with the new annotation key, perhaps here will follow some automatic placement/shifting in the future.

Best regards,
Stefan

ckuhlmann, 2017-05-06:
Hi Stefan,
thanks for your quick and helpful reply. Using your solution I was able to find `fullgeneric` for the filled-rect inductor style as well. Auto-shifting would be welcome, but this will do in the meantime. 

Do you happen to know how to set this property for individual bipoles? I tried to add a  voltage/straight label distance=0.8 or v/straight label distance=0.8 option to them but it doesn't do anything.

sistlind, 2017-05-28:
At the current git version, there is a new variable included, which can be used to set the label distance globally:
\ctikzset{bipoles/voltage/straight label distance/.initial=.5,} This value is relative to the value of bipoles/length. This means, that the absolute value is the same for all bipoles. I am not shure, if this makes sense?

sistlind, 2017-05-28:
Please reopen, if this does not solve your issue

closed 2017-05-28 by sistlind