circuitikz/circuitikz #76: bug at paths with a node as starting point

sistlind, 2017-05-19:
There is a bug, if a node is used as a starting point of a path without defining an explicit anchor:

```
\begin{circuitikz}
	\draw 
	
	%npns rechte Diffstufe
	(3,0) node[npn, anchor=C] (npn5) {}
	(npn5.B)++(1,0) node[] {$T_5$}% Beschriftung
	(4,0) node[npn, xscale=-1, anchor=C] (npn6) {}
	(npn6.B)++(-1,0) node[] {$T_6$}% Beschriftung
	(npn5.E)++(0.5,-0.5) node[npn,xscale=-1,anchor=C](npn4) {} 
	(npn4.B)++(-1,0) node[] {$T_4$}% Beschriftung
	
	%npns linke Diffstufe 
	(0,0) node[npn, anchor=C] (npn1) {}
	(npn1.B)++(1,0) node[] {$T_1$}% Beschriftung
	(1,0) node[npn, xscale=-1, anchor=C] (npn2) {}
	(npn2.B)++(-1,0) node[] {$T_2$}% Beschriftung
	(npn1.E)++(0.5,-0.5) node[npn, anchor=C](npn3) {}
	(npn3.B)++(1,0) node[] {$T_3$} % Beschriftung
	
	(npn1.C)++(0,1) node[](P3){}
	(npn6.C)++(0,1) node[](P4){}
	
	(npn1.C) to[short] (P3)
	(npn6.C) to[short] (P4)
	% weiter linke Diffstufe
	(npn1.E) -- (npn2.E) 
	(P3) to[R,l^=$R_1$,-] ++(0,2) node[vcc](){}
	(npn1.B) to[short,l_=LO-,-o] ++(-0.5,0) 
	(npn6.B) to[short,l=LO-,-o] ++(0.5,0) 
	(P4) to[R,l_=$R_2$,-] ++(0,2) node[vcc]{}
	(npn1.E)++(0.5,0) to[short,*-] (npn3.C)
	(npn3.B) to[short,l_=RF+,-o] ++(-0.5,0)
	(npn4.B) to[short,l=RF-,-o] ++(0.5,0)
	(npn5.B) -- (npn2.B)

	% weiter rechte Diffstufe
	(npn5.E) -- (npn6.E) 
	(npn5.E)++(0.5,0) to[short,*-] (npn4.C)
	
	%Verbindung zw. rechter und linker Diffstufe
	(npn3.E) -- (npn4.E)
	
	
	(npn5.C) to[short,-*] (P3)
	(npn2.C) to[short,-*] (P4)
	
	(npn2.B)++(0.2,0)to[short,-o]++(0,-0.5) node[](P5){}
	(P5)++(0,-0.3)  node[](P6){LO+}
	
	(npn3.E) ++(1.5,0) to[dcisource=$I1$,*-] ++(0,-2) node[ground](){}
	%------------------------------------------
	;
\end{circuitikz}
```

There appear anwanted lines(between emitter of T2 and R1 as well as from LO- to R2):
[image]

Rmano, 2020-07-01:
The new path fixes this. Notice that now the original example compiles like this:

[image]

Which is **correct** --- `P3` and `P4` are nodes and so they have a non-zero sizes. You can see what's happening if you add `[draw=red]` to them: 

[image]

The behavior of forcing the start point at the center of the node is not standard and creates a lot of problems.

sistlind, 2020-07-02:
Hi Rmano,
if your last screenshot is the result of the new paths, in my opinion this is a huge drawback of this code, as it will break nearly all the drawings I have done for my lectures. I think it is not convenient to always specify the center anchor. Furthermore, nodes do not need to have a border. See difference between "coordinate" and "node".

Rmano, 2020-07-02:
Hi! I understand your posiion, but the normal behavior in `tikz` is this: look:

```

Rmano, 2020-07-02:
Hi, I understand what you say, but the correct behavior in tikz is to connect the border of the nodes (even if invisible):

```latex
\documentclass[border=10pt]{standalone}
\usepackage{tikz}
\usetikzlibrary{arrows.meta,positioning,calc}
\begin{document}
\begin{tikzpicture}
    \node [](A) at (0,.2) {};
    \node [](B) at (2,.2) {};
    \node [](C) at (0,0) {};
    \node [](D) at (2,0) {};
    \draw (A) -- (B);
    \draw (C.center) -- (D.center);
\end{tikzpicture}
\end{document}
```
[image]

For this `coordinates` exists --- there are tens of post in TeX.SE about this, for example[ this](https://tex.stackexchange.com/questions/99398/why-dont-my-tikz-lines-join),

Forcing the behavior of substitute the start and end node with the center to the center of nodes has as the effect of

1. that you can't use `to[R]` (for example) between the border of two nodes, and that you have to manually select the appropriate anchor border;
2. that you can't start the path replacement code with `-- (\tikztostart)`, which will force the ugly hack for the line joins (that sometimes fails as in the case above)

So I think it's a good thing to do (I created an embedded rollback point at 1.1.2 for backward compatibility on purpose); the original code is now correctly rendered by TikZ standard (and it's easily fixable by changing `node` to `coordinate` or adding `inner sep=0pt` to the two nodes).

To explain point 1 here: this code

```latex
\begin{tikzpicture}[every node/.style={inner sep=1cm}]
    \node [draw](A) at (0,0) {A};
    \node [draw](B) at (4,2) {B};
    \draw (A) to[R] (B);
\end{tikzpicture}
```

renders as 

[image]

in 1.1.2 and previous (since 0.6, I think) 

and as

[image]

now --- you can easily have the old behavior (if you really wanted it) with using `A.center` and `B.center` explicitly, but it's really complex to obtain the new one --- which in the rest of `tikz` is considered the correct one.

Rmano, 2020-07-02:
> Here I posted some other example: https://tex.stackexchange.com/questions/360264/correct-linejoins-at-tikz-to-path

Thanks, I will study this.

Rmano, 2020-07-02:
Notice that in the example posted above now it works as should (in my opinion):

```latex
begin{tikzpicture}[very thick]
    \draw (2,2) node[draw=red](A){};
    \draw (2,2) coordinate(B);
    \draw (0,0) to[esource](0,2) to[esource](A) to[esource] ++(2,-2);
    \begin{scope}[xshift=4.5cm]
        \draw (2,2) coordinate(B);
        \draw (0,0) to[esource](0,2) to[esource] (B) to[esource] ++(2,-2);
    \end{scope}
\end{tikzpicture}
```

[image]

The line joins are correct and the landing/starting position from the node are the expected one.

commit 551a9421 (2017-05-19): Fixes #76: to-paths with a starting node are now drawn correctly
commit 6efc2ee8 (2017-05-19): Fixes #76: to-paths with a starting node are now drawn correctly
PR #418 (2020-07-01): Rewrite path formation
This basically removes the patches introduced for issue #58 and #76
    
Reverts these:
-  https://github.com/circuitikz/circuitikz/commit/6efc2ee86fb3b2b479b3039526fcc90d3f9201d2
-  https://github.com/circuitikz/circuitikz/commit/352d2bf34054d6450a0ffe92529876f4e2398ac6
    
Then, it carefully builds the anchors without moving the start of the path, and then join it with a leading `--`

Fixes #417 
Fixes #76 
Fixes #58
closed 2020-09-27 by Rmano