circuitikz/circuitikz #78: Request : Thermogenerator

arbolis, 2017-05-28:
I'd appreciate if someone could implement the thermoelectric generator symbol (can be found there : http://www.electronic-symbols.com/electric-electronic-symbols/heat-sources-symbols.htm) which looks like a box with a thermocouple and a G letter just above the thermocouple.
Thanks!

alexpacini, 2017-10-09:
I include an image of the linked symbol for reference:
[image]

Rmano, 2018-11-25:
This is a very specific symbol; maybe building it in a `tikz` `pic` would be more appropriate. Where are the terminals/leads?

Rmano, 2018-12-15:
It seems that there is no more interested in this issue; I will close it. Please re-open if you deem it necessary.

closed 2018-12-15 by Rmano