circuitikz/circuitikz #79: Undefined control sequence american resistor

greenbasilisk, 2017-06-15:
\documentclass{standalone}
    
    \usepackage{tikz}
    \usepackage{circuitikzgit}
    
    \listfiles
    
    \begin{document}
    
    \begin{tikzpicture}
      \draw (0,0) to[american resistor] (2,0);
    \end{tikzpicture}
    
    \end{document}

compiles using `lualatex` and `pdflatex` with the error

    ./tmp.tex:11: Undefined control sequence.
    \pgf@sh@bg@resistorshape ...sformationadjustments 
                                                      \pgfmathsetlength {\pgf@ci...
    l.11   \draw (0,0) to[american resistor] (2,0)

`\listfiles` output is

    standalone.cls    2015/07/15 v1.2 Class to compile TeX sub-files standalone
    ifluatex.sty    2010/03/01 v1.3 Provides the ifluatex switch (HO)
       ifpdf.sty    2011/01/30 v2.3 Provides the ifpdf switch (HO)
     ifxetex.sty    2010/09/12 v0.6 Provides ifxetex conditional
     xkeyval.sty    2014/12/03 v2.7a package option processing (HA)
     xkeyval.tex    2014/12/03 v2.7a key=value parser (HA)
    standalone.cfg    2015/07/15 v1.2 Default configuration file for 'standalone' cl
    ass
     article.cls    2014/09/29 v1.4h Standard LaTeX document class
      size10.clo    2014/09/29 v1.4h Standard LaTeX file (size option)
        tikz.sty    2013/12/13 v3.0.0 (rcs-revision 1.142)
         pgf.sty    2013/12/18 v3.0.0 (rcs-revision 1.14)
      pgfrcs.sty    2013/12/20 v3.0.0 (rcs-revision 1.28)
    everyshi.sty    2001/05/15 v3.00 EveryShipout Package (MS)
      pgfrcs.code.tex
     pgfcore.sty    2010/04/11 v3.0.0 (rcs-revision 1.7)
    graphicx.sty    2014/10/28 v1.0g Enhanced LaTeX Graphics (DPC,SPQR)
    graphics.sty    2014/10/28 v1.0p Standard LaTeX Graphics (DPC,SPQR)
        trig.sty    1999/03/16 v1.09 sin cos tan (DPC)
    graphics.cfg    2010/04/23 v1.9 graphics configuration of TeX Live
      pdftex.def    2011/05/27 v0.06d Graphics/color for pdfTeX
    infwarerr.sty    2010/04/08 v1.3 Providing info/warning/error messages (HO)
     ltxcmds.sty    2011/11/09 v1.22 LaTeX kernel commands for general use (HO)
      pgfsys.sty    2013/11/30 v3.0.0 (rcs-revision 1.47)
      pgfsys.code.tex
    pgfsyssoftpath.code.tex    2013/09/09  (rcs-revision 1.9)
    pgfsysprotocol.code.tex    2006/10/16  (rcs-revision 1.4)

Using `european resistor` the code complies flawlessly. `R` is also not working.

sistlind, 2017-06-15:
I think your tikz/pgfversion is too old. Please update your tex-installation.

greenbasilisk, 2017-06-15:
Thank you, that did the job. However, you may update the manual since I had indeed a look at the requirement sections.

closed 2017-06-15 by greenbasilisk