circuitikz/circuitikz #81: European Symbols not scalable

bkab, 2017-07-02:
I noticed that european symbols for resistors and inductors (i have not tested the other ones) can not be scaled with pgfkeys (`\ctikzset{bipoles/resistor/height=.6}` for example).
Would be great, if this issue can be resolved.

sistlind, 2017-07-02:
Have you tried \ctikzset{bipoles/generic/height=.6} ?

bkab, 2017-07-02:
I tried it right away and indeed it is working with `\ctikzset{bipoles/generic/height=.6}` for resistors and `\ctikzset{bipoles/fullgeneric/height=.6}` for inductors respectively, thank you very much.

Nevertheless, in my opinion an adaption regarding the key ´\ctikzset{bipoles/resistor/height}´ would be usefull regarding consistency with US symbols.

closed 2017-07-02 by bkab