circuitikz/circuitikz #82: Lnigbt and Tnigbt issues

martinbra, 2017-07-07:
This works as expected:
(0,0) to[Tnigbt] (0,2)

But this doesn't draw anything
(1,0)	node[Tnigbt]{}

But if I try to draw the Lnigbt model, this works:
(1,0)	node[Lnigbt]{}

but this doesn't:
(0,0) to[Lnigbt] (0,2)

Using:
\pgfcircversion = git:485934d
\pgfcircversiondate = 2017/06/16

Resuming:
```
       | to cmd  | node cmd
--------------------------------
Tnigbt | works   | doesn't
Lnigbt | doesn't | works
```

sistlind, 2017-07-07:
This is the intended behaviour.
The T at the beginning indicates a "transistor-path". Without it, it can be used as a regular node.
Just try TLnigbt!

closed 2017-07-07 by sistlind

martinbra, 2017-07-08:
Thanks for suck a quick response. It worked. 
I think that the manual was misleading me, as I thought the T and L where from the gate input format "T" and "L", and not from "transistor-path".