circuitikz/circuitikz #83: Circuitikz new current source symbol problem

superashtex, 2017-07-08:
In the new version of the Circuitikz package, the American style current sources have an extra arrow on the connecting lines in addition to the arrow inside the shape (circle or diamond). The extra arrows are really not necessary because the arrow inside the current source shows the direction of the current.

The other major problem with this arrangement is that if the value of the current is a long expression (an equation for example), it will overlay the symbol. 

Please see the following figure. Any help to fix this problem would be highly appreciated.

[image]

sistlind, 2017-07-08:
Just use l=..... instead of i=.... to name the component.

superashtex, 2017-07-08:
I did. No effect. Here is the line:

to[cI_=$z_{nm12}.y_{nm11}.i_{nm22}$]

Am I missing something?
Thanks

sistlind, 2017-07-08:
Try to[cI_,l=$z_{nm12}.y_{nm11}.i_{nm22}$]

superashtex, 2017-07-08:
It changed the controlled source to independent source with the same format:

[image]

sistlind, 2017-07-08:
Are you shure? Here it is working:
```
\documentclass{standalone}
\usepackage[american]{circuitikz}
\begin{document}
\begin{tikzpicture}
\draw (0,0) to[cI_=$z_{nm12}.y_{nm11}.i_{nm22}$] (0,2)to[cI,l_=$z_{nm12}.y_{nm11}.i_{nm22}$] (0,4);
\end{tikzpicture}
\end{document}
```
leads to:
[image]

Can you provide a full MWE?

superashtex, 2017-07-08:
It works. My mistake. I thought the letter before "=" in "cI_,l=" was "I". It is in fact little "L"! It became obvious when you posted the code.

Thanks a lot for your help.

closed 2017-07-08 by sistlind