circuitikz/circuitikz #87: Flip vamp upside down

senselessDev, 2017-07-16:
Is it possible to flip all the "non-node" symbols for example upside down? I have no idea how to do it...

MWE: 
```
\documentclass{article}
\usepackage{tikz}
\usepackage[european,siunitx]{circuitikzgit}

\begin{document}

\begin{circuitikz}
  \draw
  (0,0) node[antenna, yscale=.5] (tx1) {}
  to ++(-1,0) to [vamp] ++(-2,0);
\end{circuitikz}

\end{document}
```

[image]

sistlind, 2017-07-16:
Try [vamp,mirror]

senselessDev, 2017-07-16:
Granted, stupid me. No I also found the documentation with mirror instead of flip..

But then another question popped up:

```
\documentclass{article}
\usepackage[european,siunitx]{circuitikzgit}

\begin{document}

\begin{circuitikz}
  \draw
  (0,0) node[antenna, yscale=.5] (tx1) {}
  to ++(-1,0) to [amp, t={\small VGA}] ++(-2,0)
  to [amp, invert, t={\small VGA}] ++(-2,0)
  to [amp, mirror, t={\small VGA}] ++(-2,0)
  to [amp, mirror, invert, t={\small VGA}] ++(-2,0);
\end{circuitikz}

\end{document}
```

[image]

Is it possible to get this right?

sistlind, 2017-07-16:
You can put the text in a rotatebox or you have to place the text after
drawing the symbols.

senselessDev, 2017-07-16:
Ok, so not directly via mirror-text or something like that. 
Thanks for your fast help!

closed 2017-07-16 by senselessDev