circuitikz/circuitikz #89: resistor disregards "voltage/bump b" and "voltage/european label distance"

hmenke, 2017-08-08:
The `resistor` bipole does not obey the `voltage/bump b` and `voltage/european label distance` keys (more specifically, all bipoles derived from `generic` seem to have this issue).

```latex
\documentclass{article}

\usepackage[european]{circuitikz}

\ctikzset{
  voltage/bump b/.initial=4,
  voltage/european label distance/.initial=3
}

\begin{document}

\begin{circuitikz}
\draw (0,0) to [resistor,l={$R_1$},i={$I_1$},v^={$U_1$}] (4,0);

\draw (0,2) to [capacitor,l={$C_2$},i={$I_2$},v^={$U_2$}] (4,2);
\end{circuitikz}

\end{document}
```
[image]

sistlind, 2017-08-08:
This is because there is a shape-specific definition set, which will overwrite the default value

Try this tikzset:
```
\ctikzset{
  voltage/bump b/.initial=4,
  voltage/european label distance/.initial=3,
  bipoles/generic/voltage/european label distance/.initial=4,
  bipoles/generic/voltage/bump b/.initial=5
  }
```

hmenke, 2017-08-08:
It's a bit inconvenient that this overrides the default on the "more global" scope.  It also overrides
```
\ctikzset{
  bipoles/resistor/voltage/european label distance/.initial=4,
  bipoles/resistor/voltage/bump b/.initial=5
}
```

sistlind, 2017-08-08:
But this is the normal behaviour that a more specific option overwrites a global option, isn't it?

But you are right, the tikz-keys are hard to understand, especially the european/american auto-switched parts like resistor and inductor. But this was coded along time ago and its hard to change today, due to a lot incompatibilities.

The resistor key you mentioned will not work, because it is the american-shaped resistor. The european resistor is called generic internally (don't ask...).

Nevertheless, if i find some time, i will try to document all the config-keys.

closed 2017-10-08 by sistlind