circuitikz/circuitikz #91: incorrect positioning of labels at low/high scaling factors

sistlind, 2017-08-25:
[circuitikz_beispiel.txt](https://github.com/circuitikz/circuitikz/files/1251594/circuitikz_beispiel.txt)

Rmano, 2018-11-25:
I see this: 
[image]

Is the problem the label at $R_L$?

Rmano, 2019-02-18:
I think I have found where the error is, but I can't see why it happens. I added in `pgfcirclabel.tex` a debug code to see the anchor, and the result is: 
```
\begin{circuitikz}[scale=1]
\draw (0,0) node[anchor=east] {B}
    to[short, o-*] (1,0)
    to[R=20<\ohm>, *-*] (1,2)
    to[R=10<\ohm>, v=$v_x$] (3,2) -- (4,2)
    to[cI=$\frac{\si{\siemens}}{5} v_x$, *-*] (4,0) -- (3,0)
    to[R=5<\ohm>, *-*] (3,2)
    (3,0) to[R=10<\ohm>] (1,0)
    (1,2) to[short, -o] (0,2) node[anchor=east]{A}
    ;
\end{circuitikz}

\bigskip\par\null\bigskip

\begin{circuitikz}[scale=1.2]
\draw (0,0) node[anchor=east] {B}
    to[short, o-*] (1,0)
    to[R=20<\ohm>, *-*] (1,2)
    to[R=10<\ohm>, v=$v_x$] (3,2) -- (4,2)
    to[cI=$\frac{\si{\siemens}}{5} v_x$, *-*] (4,0) -- (3,0)
    to[R=5<\ohm>, *-*] (3,2)
    (3,0) to[R=10<\ohm>] (1,0)
    (1,2) to[short, -o] (0,2) node[anchor=east]{A}
    ;
\end{circuitikz}

\bigskip\par\null\bigskip

\begin{circuitikz}[scale=2]
\draw (0,0) node[anchor=east] {B}
    to[short, o-*] (1,0)
    to[R=20<\ohm>, *-*] (1,2)
    to[R=10<\ohm>, v=$v_x$] (3,2) -- (4,2)
    to[cI=$\frac{\si{\siemens}}{5} v_x$, *-*] (4,0) -- (3,0)
    to[R=5<\ohm>, *-*] (3,2)
    (3,0) to[R=10<\ohm>] (1,0)
    (1,2) to[short, -o] (0,2) node[anchor=east]{A}
    ;
\end{circuitikz}
```

[image]

which seems a rounding error, but I am unable to find out where it came from. If someone can help...

Rmano, 2019-02-18:
This patch: 
```
diff --git a/tex/pgfcirclabel.tex b/tex/pgfcirclabel.tex
index 83c39fe..8b920a5 100644
--- a/tex/pgfcirclabel.tex
+++ b/tex/pgfcirclabel.tex
@@ -89,6 +89,7 @@
 \def\pgf@circ@drawlabels#1{
        \pgfextra{
                \pgfmathsubtract{\pgf@circ@direction}{90}
+                \pgfmathround{\pgfmathresult}
                \edef\pgf@circ@labanc{\expandafter\pgf@circ@stripdecimals\pgfmathresult\pgf@nil}
                
                \edef\pgf@circ@temp{\ctikzvalof{bipole/#1/position}} 
@@ -96,6 +97,10 @@
                        \pgfmathadd{\pgf@circ@labanc}{180}
                        \edef\pgf@circ@labanc{\expandafter\pgf@circ@stripdecimals\pgfmathresult\pgf@nil}
                \fi
+                \ifnum\pgf@circ@labanc > 180
+                    \pgfmathsubtract{\pgf@circ@labanc}{360}
+                    \edef\pgf@circ@labanc{\expandafter\pgf@circ@stripdecimals\pgfmathresult\pgf@nil}
+                \fi
                \ifpgf@circuit@bipole@inverted
                        \pgfmathadd{\pgf@circ@temp}{180} %If shape is inverted, use opposite anchor
                        \edef\pgf@circ@temp{\expandafter\pgf@circ@stripdecimals\pgfmathresult\pgf@nil}
``` 
seems to fix/workaround it.

Rmano, 2019-02-18:
...but it does not solve the initial @sistlind  problem. Will look a bit more into it, but label positioning is somehow flaky...

Rmano, 2019-02-19:
Trying a patch, at least it seems better. Please check PR #155

Rmano, 2019-02-20:
I have found a better way. With the last commit on PR #155, I have now the following, which seems to fix everything. I will run the manual with it, and if it's ok, I'll commit it. 

[image]

Rmano, 2019-02-21:
See discussion on pr #155; reopened

Rmano, 2019-02-21:
Ok, the problem is tough, I think. I tried to revert to the previous state, but it seems that the shift of (1ex) is badly calculated when `scale` is concerned. I added code to show the y-coordinate  of the label anchor before and after the shift, and I see this: 
1) no scale: moved up 3.3pt, (0.75ex)
[image]
2) `scale=2`: moved up the same 3.3pt , so it has been not scaled. Not sure if it's ok. 
[image]
3) your scale of `yscale=0.012`: moved **down** 0.2pt? I suspect a TeX numeric overflow here. 
[image]

I will try to prepare am MWE and asking around...

Rmano, 2019-02-21:
Ok, peskier and peskier. See https://tex.stackexchange.com/questions/475976/using-font-relative-distances-in-tikzpictures --- the result with the extreme scale is completely different. Something is going on in the coordinate system here.
I think I will anyway revert the anchor change, @sistlind  is right, it is unexpected and visually strange.

Rmano, 2019-02-21:
@sistlind , I created PR #157 following Eric's idea in https://tex.stackexchange.com/a/476004/38080, and it seems to work for all my cases.

Rmano, 2019-02-23:
I think that #157  is ready to go...

Rmano, 2019-02-27:
Fixed in #157; closing issue.

commit 6e208c50 (2019-02-20): Merge pull request #155 from Rmano/fixlabelpos

Fix label position; fixes #91
closed 2019-02-27 by Rmano