circuitikz/circuitikz #92: Circuit elements displayed incorrectly

hjmjohnson, 2017-09-04:


hjmjohnson, 2017-09-04:
NOTE:  Issue #69 and commit ad8c384bcc9c1c4444c31bf43631f78e5ca3a187 change the backwards behavior of how circuits are drawn.

hjmjohnson, 2017-09-04:
These two commits also break backward compatiblity with verison v0.5

commit 6a138ece477f2368bbdffb0617d2f2bdeca985ff (HEAD)
Author: Stefan Lindner <stefan.lindner@gmail.com>
Date:   Sat Apr 22 10:12:17 2017

    update documentation

commit 6524f2201368a3b2c7c570c39e7765a5b87b99fa (refs/bisect/bad)
Author: Stefan Lindner <stefan.lindner@gmail.com>
Date:   Sat Apr 22 09:16:15 2017

    current label now independent of regular label at current sources; code cleanup at current label drawing

hjmjohnson, 2017-09-04:
Older documents compile with the new package, but give the wrong rendered results as compared to version 0.5

sistlind, 2017-10-08:
This change was intended, because there was no rule in which direction voltages and currents are drawn. Therefore, it was some kind of trial and error at each symbol, which also changes, is the packet options were changed from european to american.

For the drawing direction you can use the package option "oldvoltagedirection", but there is no option for the labelplacement.

For compatibility reasons at older projects, you can use the released git-versions from here: https://circuitikz.github.io/circuitikz/

closed 2017-10-08 by sistlind