circuitikz/circuitikz #95: Issues with switches

thejoker2, 2017-11-01:
Hello,

I have issues with switches.
I cannot draw a normal switch without any arrow on it.

And the codes in the documentation don't give the expected symbols.

"**spst**" gives: 
[image]

"**ospst**" gives: 
[image]

"**cspst**" gives: 
[image]

with is not what is expected.
And there is no way to get a "normal" switch without any arrow.

Thanks for your help!

thejoker2, 2017-11-03:
I am the only one the face this issue?

thejoker2, 2017-11-05:
Any help please?

thejoker2, 2017-11-14:
:( .....

Rmano, 2017-11-14:
I sympathize... but adding an element to circuitikz is not easy. You can do it with pure tikz, like 

```
 \draw (0,0) -- ++(0.2,0) 
        node (temp) [draw, inner sep=1pt, circle, fill=white]{} 
        ++(0.6,0) node [draw,inner sep=1pt, circle, fill=white]{} 
        -- ++(0.2,0);
    \draw [thick] (temp) -- ++(0.6, 0.3);
```

[image]

(maybe in a macro, or in a `pic` environment).

thejoker2, 2017-11-14:
Thank you for your answer. :)

But what does the code in the Circuitikz documentation not give the expected results about switches?

I am the only one facing this issue?

sistlind, 2017-11-25:
Sorry for the late response :/
Please have a look at http://circuitikz.github.io/circuitikz/circuitikzmanualgit.pdf on page 19. There you find a few differen switching shapes. perhaps you are searching for the node "nos"?

thejoker2, 2018-01-03:
I tried:
`\draw (v1) to [nos] (3,2.1) to [R,l_=$\frac{R}{2}$,i=$i_2$] (v2);`
But I got this error:

> ! Package pgfkeys Error: I do not know the key '/tikz/nos' and I am going to ig
nore it. Perhaps you misspelled it.
See the pgfkeys package documentation for explanation.

sistlind, 2018-02-10:
Your code works here, please update your tex installation

closed 2018-02-10 by sistlind