circuitikz/circuitikz #96: No galvanometer or the ability to rotate components

musama95, 2017-12-07:
Why is there no component as basic as a galvanometer? I can't seem to easily rotate any of the components either.

musama95, 2018-01-18:
It's been about a month with no response from anyone.

sistlind, 2018-02-10:
I think there is no galvanometer, because nobody created some yet. Just code it, i will include it.

Please have a look at the manual, there are the keys mirror/invert which can be used to rotate components.

closed 2018-02-10 by sistlind

musama95, 2018-02-26:
Could you quote the place in the manual? I read almost all of it but I can't seem to find the inversion code. It would be really helpful! As for the galvanometer, I could make one on Illustrator, but making it in code would be impossible for me.

sistlind, 2018-02-27:
http://circuitikz.github.io/circuitikz/circuitikzmanualgit.pdf page 43