circuitikz/circuitikz #97: Conflicting with IECEE symbols

Dxhs, 2018-02-26:
Why isn't the ground symbols designed according to IEC 60417 ground symbols?

sistlind, 2018-02-27:
Could you provide a picture? I will add the symbol

Rmano, 2018-02-27:
https://www.iso.org/obp/ui#iec:grs:60417:5017 
;-)

sistlind, 2018-02-27:
Thia one ahould be included, pla have a look at the manual. Afaik it is
called rground

Dxhs, 2018-02-27:
Not a problem in itself, but a noticeable detail.
[image]

References: 
https://www.iso.org/obp/ui#iso:pub:PUB400008:en
CircuiTikz manual (2016)

Dxhs, 2018-02-27:
Geometry at the bottom: https://incompliancemag.com/article/the-grounding-symbols/

sistlind, 2018-04-25:
Sorry for the delay, the new symbols are now available at the git version. The line thickness of the symbols and the connection can be adjusted indiviually using:
`\ctikzset{monopoles/ground/connectionthickness/.initial=1}`
`\ctikzset{monopoles/ground/thickness/.initial=2}`

closed 2018-04-25 by sistlind