circuitikz/circuitikz #98: Trivial Question

circuitikzuser, 2018-03-08:
I am an old hand at LaTeX, but new to circuitikz. I read the very well written manual, and find  circuitikz to be one of the most useful packages for LaTeX.

I have unfortunately a trivial issue I cannot reolve. Upon creating the headers as far as I could figure correctly from other examples, and then running the code as in the examples in the manual, I get a problem with the interpretation of component properties.

e.g.  the built in units are not interpreted by  circuitikz package.
I am breaking my head about this simple issue, but I cannot find the reason for this in the manual or online or in the archives.  I just cannot figure what is missing and why units are not  compiled.

See below from the compiler output.

! Undefined control sequence.
\milli ->\ERROR 
                
l.77 ...2) -- (0,3) to[L, l=2<\milli\henry>] (5,3)
                                                   -- (5,2)
! Undefined control sequence.
\henry ->\ERROR 

Thanks

sistlind, 2018-04-25:
Regarding Question 1: Sorry, i am not familiar with SIunits.
Question 2: You are right, there is no such symbol. Maybe node[nigfete,nogate] helps you?

Rmano, 2018-11-25:
I will try to have a look at the SIunits thing, although I normally use `\SI{}{}` which works everywhere and has no problem. 
Can you give an example of opto-fet?

Rmano, 2018-12-15:
@circuitikzuser , your code works ok with the current git version of `circuitikz`: 

[image]

Rmano, 2018-12-15:
For the OptoFET, you have plenty of arrows in `arrows.meta` library of Tikz. For example:

```latex
\usetikzlibrary{arrows.meta}
\ctikzset{tripoles/mos style/arrows,}
\begin{circuitikz}[ 
    ]
    \draw (0,0) node[nmos,nogate](M1){};
    \draw[thin, -Latex] (M1.G) ++(-0.3,0.4) -- ++(0.6,-0.5);
    \draw[thin, -Latex] (M1.G) ++(-0.3,0.6) -- ++(0.6,-0.5);
\end{circuitikz}
```
results in 

[image]

Rmano, 2018-12-15:
I think I can close this issue. If you think it's not solved, feel free to re-open it.

closed 2018-12-15 by Rmano