circuitikz/circuitikz #103: Request: Support to draw MIGFETs (custom transistors)

merkste, 2018-06-25:
I'd like to draw field effect transistors with multiple gates, e.g., 4 gates.  The style should be similar to `node[nmos]{}` or `node[pmos,emptycircle]{}`. Is there a way to achieve this with the current package or does is require a new node type, lets say quadpoles or quintpoles?

sistlind, 2018-06-25:
Hi,
you can try the "nogate" option. This removes the regular gate from the
fets and allows you to draw individually several single gates.
Please have a look at the manual, where the option is explained.

Best regards
Stefan

Rmano, 2019-02-15:
I think that @sistlind gave the correct answer here, it's too difficult to add all the cases, and this is easily fixed. Will close, thanks,

closed 2019-02-15 by Rmano