circuitikz/circuitikz #105: Request: multiple inputs for logic gates

TheTeXnician, 2018-09-13:
`circuitikz` provides means to draw logic gates (ports). Unfortunately, it is restricted to two inputs per gate. As it is handy (from time to time) to have more than two inputs, it would be nice if this could be added.

For reference, this has been asked at TeX.StackExchange: https://tex.stackexchange.com/questions/372993/is-it-possible-to-implement-multiple-input-logic-ports-with-circuitikz

The linked post contains two answers describing possible ways to adapt the American as well as the European gates.

Rmano, 2018-11-04:
I can try to prepare a pull request if I have the ok by the author (John Kormylo)...

Rmano, 2018-11-25:
...but it's not trivial, it will be a quite big change... and I never use logic gates, so I'll need help at least to check functionality.

TheTeXnician, 2018-11-25:
@Rmano Concerning the size of the change I would expect it to be about the size of the TeX.SX answer linked. However, I would be willing to test and check the functionality.

Rmano, 2018-11-26:
@TheTeXnician  thanks. I  will try to prepare a pull request when I can, now I am a bit swamped...

Rmano, 2019-02-20:
I am trying to do it, but I have stumbled upon a problem. John Kormylo code works ok "standalone", but when I used it to substitute the code in the `pgfcirctripoles.tex` I have errors, the anchors are not correctly created. 
I will try again tomorrow, and maybe post a "failing PR" to see if anyone can help me.

Rmano, 2019-02-21:
I found the problem. The problem is that all is well for `american and port` but, if you use the shorthand `and port`, you have: 
`Package PGF Math Error: Unknown function `in' (in 'in 1').`

And this is due to the following code, 
which for whatever reason, do the port style-switch in this way (`pgfcircpath.tex`):

```
 \def\inheritlogicport#1#2{
	\pgfdeclareshape{#2 port}{
		\inheritsavedanchors[from=#1 #2 port]
		\inheritbackgroundpath[from=#1 #2 port]
		\inheritanchor[from=#1 #2 port]{in}
		\inheritanchor[from=#1 #2 port]{in 1}
		\inheritanchor[from=#1 #2 port]{in 2}
		\inheritanchor[from=#1 #2 port]{out}
		\inheritanchor[from=#1 #2 port]{center}
		\inheritanchor[from=#1 #2 port]{left}
	}
}

\ifpgf@circuit@europeanlogicport
	\inheritlogicport{european}{and}
	\inheritlogicport{european}{or}
	\inheritlogicport{european}{xor}
	\inheritlogicport{european}{not}
	\inheritlogicport{european}{nand}
	\inheritlogicport{european}{nor}
	\inheritlogicport{european}{xnor}
\else
	\inheritlogicport{american}{and}
	\inheritlogicport{american}{or}
	\inheritlogicport{american}{xor}
	\inheritlogicport{american}{not}
	\inheritlogicport{american}{nand}
	\inheritlogicport{american}{nor}
	\inheritlogicport{american}{xnor}
\fi
```

...and clearly the code fails to copy the generated anchors (even the `in 1` and `in 2`) to the "short name port". Any suggestion?

Rmano, 2019-02-21:
...and notice that it is not only a problem due to the space, because the `out` anchor is inherited, but it comes out wrong... 
```
\draw 
  (0,0) node[and port, ] (myand)  {}
  % (myand.in 1) node[anchor=east] {1}
  % (myand.in 2) node[anchor=east] {2}
  (myand.out) node[anchor=west] {3}
;
```
[image]

Rmano, 2019-02-22:
@mredaelli @sistlind @der-stefan , anyone knows why to switch gate with the style is done by inheritance and not with 
```
\tikzset{and port/.style={shape=american and port}}
```
...which seems to work?

Rmano, 2019-02-22:
I have "test" PR in #158. Seems to work here, you can check with 
```
\documentclass[border=10pt]{standalone}
\usepackage[siunitx,
    % europeanport,
    ]{circuitikz}
\def\coord(#1){node[circle, red, draw, inner sep=1pt,pin={[red, overlay, inner sep=0.5pt, font=\tiny, pin distance=0.1cm, pin edge={red, overlay,}]45:#1}](#1){}}
\begin{document}
\begin{circuitikz} [american]
\draw
  (0,0) node[and port, ] (myand)  {}
  (myand.in 1) \coord(1)
  (myand.in 2) \coord(2)
  (myand.out) \coord(3)
  (myand.north east) \coord(ne);
;
\draw
  (3,0) node[european and port, ] (myeand)  {}
  (myeand.in 1) node[anchor=east] {1}
  (myeand.in 2) node[anchor=east] {2}
  (myeand.out) node[anchor=west] {3}
;
\draw
  (0,3) node[american and port, number inputs=5] (myand)  {}
  (myand.in 1) node[anchor=east] {1}
  (myand.in 2) node[anchor=east] {2}
  (myand.out) node[anchor=west] {3}
;
\draw
  (3,3) node[european and port, number  inputs=5] (myeand)  {}
  (myeand.in 1) node[anchor=east] {1}
  (myeand.in 2) node[anchor=east] {2}
  (myeand.out) node[anchor=west] {3}
;
\end{circuitikz}
\end{document}
```

[image]

TheTeXnician, 2019-02-22:
Just for testing: A ConTeXt MWE that should also work.

```
\usemodule [circuitikz]
\def\coord(#1){node[circle, red, draw, inner sep=1pt,pin={[red, overlay, inner sep=0.5pt, font=\tfxx, pin distance=0.1cm, pin edge={red, overlay,}]45:#1}](#1){}}
\startTEXpage
\startcircuitikz[american]
\draw
  (0,0) node[and port, ] (myand)  {}
  (myand.in 1) \coord(1)
  (myand.in 2) \coord(2)
  (myand.out) \coord(3)
  (myand.north east) \coord(ne);
;
\draw
  (3,0) node[european and port, ] (myeand)  {}
  (myeand.in 1) node[anchor=east] {1}
  (myeand.in 2) node[anchor=east] {2}
  (myeand.out) node[anchor=west] {3}
;
\draw
  (0,3) node[american and port, number inputs=5] (myand)  {}
  (myand.in 1) node[anchor=east] {1}
  (myand.in 2) node[anchor=east] {2}
  (myand.out) node[anchor=west] {3}
;
\draw
  (3,3) node[european and port, number  inputs=5] (myeand)  {}
  (myeand.in 1) node[anchor=east] {1}
  (myeand.in 2) node[anchor=east] {2}
  (myeand.out) node[anchor=west] {3}
;
\stopcircuitikz
\stopTEXpage
```

Rmano, 2019-02-22:
I have 
```
tex error       > tex error on line 30 in file /home/romano/software/latex/source/test/context-test.tex: ! Misplaced alignment tab character &

<recently read> &
 
\pgf@sh@bg@european and port ...in } \pgftext {$\&
                                                  $} \pgfsetlinewidth {2\pgf...
<argument> \csname pgf@sh@bg@\tikz@shape \endcsname 
                                         \ifpgflatenodepositioning \pgfsyssoftp
\pgfutil@secondoftwo #1#2->#2
                             
<argument> ...ish }\global \pgflinewidth =\pgfutil@tempdima }
                                                  \pgfutil@ifundefined {pgf@...
\pgfutil@secondoftwo #1#2->#2
                             
...
l.30 \stopcircuitikz
```

which I do not understand... the "&" seems correctly escaped, no?

TheTeXnician, 2019-02-22:
@Rmano I cannot reproduce. For me the above example as well the context-circuitikz.tex document from the doc folder work okay (with ConTeXt MkIV from TL2018 as well as the latest beta).

Rmano, 2019-02-22:
Hmmm... Could be this? https://tex.stackexchange.com/questions/101910/ampersand-in-math-in-context.

Anyway, \& in math mode has little sense....

Rmano, 2019-02-22:
This fixed it for me: 
```
diff --git a/tex/pgfcirctripoles.tex b/tex/pgfcirctripoles.tex
index 33aac50..3de5778 100644
--- a/tex/pgfcirctripoles.tex
+++ b/tex/pgfcirctripoles.tex
@@ -931,11 +931,11 @@
       }
     }
 }
-\pgfcircdeclareeurologicport{and}{$\&$}{\pgf@circ@res@count}{}
+\pgfcircdeclareeurologicport{and}{\&}{\pgf@circ@res@count}{}
 \pgfcircdeclareeurologicport{or}{$\ge 1$}{\pgf@circ@res@count}{}
 \pgfcircdeclareeurologicport{xor}{$=1$}{\pgf@circ@res@count}{}
 \pgfcircdeclareeurologicport{not}{$1$}{1}{not}
-\pgfcircdeclareeurologicport{nand}{$\&$}{\pgf@circ@res@count}{not}
+\pgfcircdeclareeurologicport{nand}{\&}{\pgf@circ@res@count}{not}
 \pgfcircdeclareeurologicport{nor}{$\ge 1$}{\pgf@circ@res@count}{not}
 \pgfcircdeclareeurologicport{xnor}{$=1$}{\pgf@circ@res@count}{not}
```

Rmano, 2019-02-22:
Ok, I think I have it more or less shaped up... tomorrow I will try to add the documentation, and then tidy up the PR trying my best to preserve authorship.

Rmano, 2019-02-23:
Now the update of the manual is in. I think that the PR is in good shape for merging.

Rmano, 2019-02-27:
Branch merged; closing issue.

closed 2019-02-27 by Rmano