circuitikz/circuitikz #130: Request : new Bulb symbol

WoolSeo, 2018-11-21:
Hi, everyone.
Circuitikz and this site  are very nice to me. I'm physics teacher.

There is lamp symbol. but I need bulb symbol.

Could you make bulb symbol for physics teacher?
like https://www.howequipmentworks.com/wp-content/uploads/2015/01/bulb_symbol.gif

Rmano, 2018-11-21:
Like this one? 

```
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, oldvoltagedirection]{circuitikzgit}
% needs the latest version of circuitikz, see github (dunno why)
% \usepackage[siunitx]{circuitikz}
\makeatletter
\ctikzset{bipoles/bulb/height/.initial=.8}
\ctikzset{bipoles/bulb/width/.initial=.8}
% #1 - additional anchors
% #2 - lower y-size of the bipole (from the center).
% #3 - #shape is the name of the shape
% #4 - upper y-size of the bipole (from the center)
% #5 - width of the bipole
% #6 - macros drawing the bipole
\pgfcircdeclarebipole
{}
{\ctikzvalof{bipoles/bulb/height}}
{bulb}
{\ctikzvalof{bipoles/bulb/height}}
{\ctikzvalof{bipoles/bulb/width}}
{%
    \pgfsetlinewidth{\pgfkeysvalueof{/tikz/circuitikz/bipoles/thickness}\pgfstartlinewidth}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{\pgf@circ@res@zero}}
    \pgfpathlineto{\pgfpoint{0.4\pgf@circ@res@left}{\pgf@circ@res@zero}}
    \pgfpatharc{0}{-180}{0.4*\pgf@circ@res@left}
    \pgfsetbeveljoin
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@zero}}
    \pgfpathellipse{\pgfpointorigin}{\pgfpoint{0}{0.8\pgf@circ@res@up}}{\pgfpoint{0.8\pgf@circ@res@left}{0}}
    \pgfusepath{draw}
}
%% Activate the bipole
\def\pgf@circ@bulb@path#1{\pgf@circ@bipole@path{bulb}{#1}}
\compattikzset{bulb/.style = {\circuitikzbasekey, /tikz/to path=\pgf@circ@bulb@path, l=#1}}
%
\makeatother
\def\coord(#1){coordinate(#1)}
\def\coord(#1){node[circle, red, draw, inner sep=1pt,pin={[red, overlay, inner sep=0.5pt, font=\tiny, pin distance=0.1cm, pin edge={red, overlay,}]75:#1}](#1){}}
\begin{document}
\begin{circuitikz}[ american, ]
    %
    \draw (0,0) to[bulb] ++(2,0);
    \draw (0,1) to[bulb, color=red, bipoles/thickness=1] ++(3,0)
    to[bulb, color=blue, mirror] ++(0,-2);
    
\end{circuitikz}
\end{document}
```

[image]

Rmano, 2018-11-25:
Hmmm.... it's ok for you?

commit 0cbd16ed (2018-12-05): Merge pull request #142 from Rmano/bulb

Added the bulb symbol, fixes #130
closed 2018-12-05 by Rmano