circuitikz/circuitikz #131: Maintainer needed?

sistlind, 2018-11-22:
Hi,
as I am currently not use circuitikz regularly (phd done) and this seem to be true also for the other two maintainers, we should think of a new/additional developer.

@der-stefan @mredaelli  please comment about your further plans and what you think about this.

If someone is interested in officially supporting this project including regular releases, please contact us.

Regards Stefan

Rmano, 2018-11-22:
I could try giving it a shot if the original authors are willing to help me about a couple of things. The positive side is that I am using it quite a lot; the negative side is that a) I know nothing of ConTeXt nor b) how to interact with CTAN, and that part of the code is still a mystery for me... 

And we have to agree about how to sort out the voltage thing ;-)

mredaelli, 2018-11-22:
Hey!

First of all, thanks once again to the Stefans for shouldering the task of digging through my horrible TeX code, and for adding so much to it.
I took a long vacation from this library, mostly because I'm not using it anymore (also because most of the work that was done is in electronic components, and I love Circuit Theory but I dearly hate Electronics...), and I might be able pitch in again, assuming there's someone who leads :)

So, I think that having @Rmano take over would be great (the idea to be in Italian hands is also appealing, not for nationalistic reasons of course but because he seems to follow and understand the same graphical conventions I had in mind when I implemented the whole thing :) ), and I'd be happy to help a little.

der-stefan, 2018-11-22:
Hi others,
I fully agree that we should open the project and rise @Rmano to the maintainer. @sistlind was the maintainer after @mredaelli - I just contributed some symbols and functions, but never maintained the package... So, welcome on board, Romano!

Rmano, 2018-11-23:
So ok --- thanks. I will need help if the ConTeXt part broke, for sure, and on the CTAN process. I'll put here a kind of release plan during the weekend, and propose a fix (I have a couple of ideas) for the voltage directions tangle. 
What are the following steps? I imagine I'll need write access to the repository.

sistlind, 2018-11-24:
Hi,
@mredaelli: welcome back :)
@Rmano: I think none of the maintainer is using context, so lets keep the fingers crossed that there will never be an issue here. The CTAN process is really complicated and confusing, therefore, I highly automated it. If you have a linux machine or cygwin it is as easy as "make ctan" IIRC.

I will write an email in the next days describing it a bit more.

I will try to help where I can and even fix some bugs, but I see not really a chance to maintain this package anymore, because I do not use it regularly enough.

I think we can close here and going on per mail. I just want to highlight this topic here, otherwise some guys could think the package is not maintained anymore.

closed 2018-11-24 by sistlind