circuitikz/circuitikz #135: Makefile cleanup

mredaelli, 2018-11-26:
I was looking around the repository, and I was wondering, @sistlind, if there is (still?) a reason for having both the `manual-git` and `manual-latex` targets.

I would say we need only one (using the `circuitikz` version in the checked-out repository), but before deleting by all means let me know what you think.

Rmano, 2018-11-26:
I agree that the Makefile would be improved by a cleanup ;-). In `develop` branch, the `manual-latex` is built using the version in the checked-out repository, by using `TEXINPUTS` environment variable.

sistlind, 2018-11-26:
One is for the gitversions and one for official releases. Maybe this can be simplified, but please check if the automatic travis builds still work. IIRC the git target compiles the corresponding manual for each version, which is available here https://github.com/circuitikz/circuitikz/tree/gh-pages

mredaelli, 2018-11-27:
I see, great! Thanks :)

Rmano, 2018-11-27:
I declare I do not know anything about this deploying thing. where can I read about it? Thanks!

sistlind, 2018-11-27:
Each commit triggers a travis build, which mainly consists of a "make manual-git".
If the commit was in the master branch, the generated file and a "git-version" is automatically placed at the github-pages site.

See config: https://github.com/circuitikz/circuitikz/blob/master/.travis.yml
And script, which is run after build: https://github.com/circuitikz/circuitikz/blob/master/tools/deploy.sh

Rmano, 2018-12-03:
@sistlind  I merged a couple of pull requests (the simplest ones), but I do not see the new builds. Should I do something special like merging in gh-pages or similar things? 
Thanks

sistlind, 2018-12-04:
@Rmano : the build was complete but could not checked in, because the deploy key was missing. No idea why? Nevertheless, I added it again, now it should work. I retriggered the last build, see here the results: https://circuitikz.github.io/circuitikz/

@mredaelli : the current travis.ci implementatio is deprecated and should be replaced by the travis-ci app. I have installed this, but cannot activate it, because I am not admin of this repository. Could you have a look at the settings->integration->apps->travis ci and configure/activate it? Currently everything is running over the old interface "services" but this will stop working a the end of the year

Rmano, 2018-12-04:
Thanks! 
Ok, it seems I have to try to get up to speed with this Travis and deploy thing (not to mention the CTAN process). I'll search around, but if you can point to some starting-level documentation it will be greatly appreciated.

Thanks again!

mredaelli, 2018-12-04:
I took a look, but it seems to be already configured (and you appear as Organization owner in the GitHub App page).

Well, if it stops at the end of the year, we will fix it :)

sistlind, 2018-12-04:
I have migrate it to the new app interface. Lets check at the next push/pull request :)

@Rmano: travis just performs an make manual-git and calls afterwards /tools/deploy.sh. You may have a look at this file and .travis.yml to start exploring it.

sistlind, 2019-02-24:
Hi Rmano,
it pulls the master branch and builds it (make git-manual). In case of success it pulls the gihub-pages branch and place the new doc.pdf and circuititkz-git.sty in the correct places, commit the changes and push it.
For this push a api-key is necessary, which is encrypted by travis. This way the whole scripts can be within the public github repository without given somebody a pushable api key.  I can send you the private key which I used to generate this encrypted keys, just contact me if you need it. (But you can also generate your own.)

Regards Stefan

Rmano, 2019-02-24:
Hi Stefan, 
thanks --- if you could just describe the thing (I am a bit lost) I will be happy to generate my own key or using yours, if needed (now it's ok, so I will not touch it). And if you can describe the procedure to do a CTAN release, I will be delighted... I'd like to go with a 0.9.0 release as soon as all the reasonable issues are closed.

Rmano, 2021-01-05:
I think that with https://github.com/circuitikz/circuitikz/pull/481/commits/ac7700e97a71732fd70a4b67ea70223b849957a7 I can close it... ;-)

closed 2021-01-05 by Rmano