circuitikz/circuitikz #143: Problem in processing `l=` (and similar) keys

Rmano, 2018-12-11:
Consider the following piece of `tikz` code: 
```
\draw (0,0) node[node contents=$a=b$]{};
``` 
...it will fail, because the parser for the keys will get confused by the `=` symbol. So far, so good; you can adjust that "hiding" the `=` into braces: 
```
\draw (0,0) node[node contents={$a=b$}]{};
```
will work. Unfortunately, in `circuitikz`, `l={$R=R_1$}` will fail with error: 
```
Extra }, or forgotten $. 
Missing $ inserted.      
Missing } inserted.      
```
...and a fun rendering involving `\pgfextra`. Workarounds are possible, but they are ugly. 
```latex
Complete test code: 
\documentclass[border=10pt]{standalone}
%\usepackage[siunitx]{circuitikz}
\usepackage[siunitx]{circuitikzgit}
\begin{document}
\begin{circuitikz}[
    ]
    % you normally should hide "=" and "," and "[]" in keywords
    \draw (0,0) node[node contents={$a=b$}]{};
    % this fails (shouldn't)
    \draw (0,-2) to[R, l={$R=R_1$}] ++(3,0);
    % this works (but spacing is wrong)
    \draw (0,-4) to[R, l=$R{=}R_1$] ++(3,0);
    % this is correct (but it's a pain)
    \draw (0,-6) to[R, l=$R{{}={}}R_1$] ++(3,0);
\end{circuitikz}
\end{document}
```

and result (cropped): 

[image]

sistlind, 2018-12-11:
This is an old bug :/ IIRC this is not solvable whitin circuitikz and it should be already documented somewhere in the manual.

Rmano, 2018-12-15:
Best workaround is `\newcommand{\eq}{=}` and use `\eq` instead of `=` when needed. Will try to add it to the manual...

sistlind, 2018-12-15:
good idea, but we should maybe use a more unique cmd like \ctikzeq

Rmano, 2018-12-15:
> good idea, but we should maybe use a more unique cmd like \ctikzeq

Well, I was thinking of giving the hint in the manual and let the user choose it's "equivalent" ... I would not use `\ctikzeq` really, `\eq` is easier and more intuitive, but I agree that we can't use it in the package.

sistlind, 2018-12-15:
I am just a  bit afraid that it could collide with other packages using
this macro.Just because it is so intuitive and good abbreviation for equal
as well as equation.

Rmano, 2018-12-15:
> I am just a bit afraid that it could collide with other packages using this macro.Just because it is so intuitive and good abbreviation for equal as well as equation.

Yes, you are right. I think it's better not to add anything to `circuitikz`, and letting the user decide. I'll prepare a patch for the user manual.

commit 32bc726c (2019-02-17): Merge pull request #154 from Rmano/workaround143

Warn about "=" and "," in labels. Fixes #143
closed 2019-02-17 by Rmano