circuitikz/circuitikz #146: Problems combining Circuitikz with Tikz pics

qoheniac, 2019-01-24:
If I take working code and put it inside a tikz pic, it sometimes works with circuitikz and sometimes it doesn't. See the minimal example below for example: The line with P1 is working while the one with P2 isn't.
```
\documentclass{article}
\usepackage{circuitikz}
\begin{document}
  \ctikzset{bipoles/length=1.5em}
  \tikzset{
    block/.style = {draw, thick, rectangle, minimum height=3em, minimum width=3em},
    tran/.pic = {
      \draw (0, 0) node[transformer] (s) {};
      \draw ($.25*(s.A1)+.25*(s.A2)+.25*(s.B1)+.25*(s.B2)$) node[block] (b) {};
    },
    rect/.pic = {
      \draw (0, 0) node[block] (b) {};
      \draw (b.west) to[diode] (b.north);
    }
  }
  \begin{figure}[h]
    \centering
    \begin{tikzpicture}
      \pic (P1) at (0, 0) {tran};
      \pic (P2) at (3em, 0) {rect};
    \end{tikzpicture}
  \end{figure}
\end{document}

```

Rmano, 2019-01-25:
Yes --- the elements used as `to` not only build a path, but they use them. So probably there is a bad interaction here with the path-manipulation that the `pic` is doing. 
  
It's probably implicit in the design; I do not think it's easily fixable.

qoheniac, 2019-01-25:
Oh, too bad. I really like pic, but then I'll just use loops to repeat parts as usual. Thanks for your fast reply.

Rmano, 2019-02-15:
I will mark this as "couldn't fix". If someone can have an idea on how to workaround this thing, I will be happy to oblige... but I suspect it would be a LOT of work (probably separating the components in "backgroundpath" and "behindbackground path" and doing the filling there could help... do anybody treid it?

Rmano, 2019-02-16:
Will close this as too complex for now; thank you for the comment and do not hesitate to write again if you need it.

closed 2019-02-16 by Rmano