circuitikz/circuitikz #151: Request: Any method to acquire the coordinate of anchor point?

ZackhamGuo, 2019-02-16:
Some time I may want to draw something exactly on the up/down/left/right of the anchor point of a tripole/monopole, for instance something like:
([coordinate x of my and port.in 1 ], 0) -- (my and port.in 1)
Is there any method of achieving this?

Rmano, 2019-02-16:
I am not sure I understand exactly what you need... this thing? Notice that this is not a circuitikz thing, it's a generic `TikZ` coordinate manipulation: 

```
\begin{circuitikz}[
    ]
    \draw (0,0) node[and port](P){};
    % relative coords:
    \draw (P.in 1) to[short, -*]  ++(0,1) coordinate(A);
    % using vertical|horizontal crossing; (node1 -| node2)
    % means the point with x coord of node1 and y coord of node2
    \draw (P.out) -- (A -| P.out);
\end{circuitikz}
```
[image]

For more complex things, you can extract x and y of any point in TikZ with the `let` primitive: https://tex.stackexchange.com/questions/66125/extract-x-value-from-coordinate-in-tikz

Rmano, 2019-02-16:
I am closing this now, please reopen if I misunderstood you. Thanks for using circuitikz!

closed 2019-02-16 by Rmano