circuitikz/circuitikz #161: req: white & pink noise signal generator

mpancorbo, 2019-03-03:
As for version 0.8.3. no symbols for white & pink noise signal generators are shipped (as far as I saw). May you consider such devices.

If you have suggestions about how to build them with actual symbols I will read read them thankfully.

Rmano, 2019-03-03:
Hmmm.... I worked in electronic noise quite a lot, and the generators I used where the ones that I added in the git version (and that will be in the next release, but you can use them using the git version now): see merged PR https://github.com/circuitikz/circuitikz/pull/136

Rmano, 2019-03-03:
I mean, I always used generic noise sources and then marked the name and the appropriate PSD in the text. I am not aware of specific symbols for white or pink or colored noise sources.

mpancorbo, 2019-03-03:
Thanks for the tip! I'll take a look to the git version.

> I mean, I always used generic noise sources and then marked the name and the appropriate PSD in the text. I am not aware of specific symbols for white or pink or colored noise sources.
> But if I am wrong, please point me to a source for them.

In fact, it is the first time I must face a noise generator in a circuit, so I can't point to better symbols.

closed 2019-03-03 by mpancorbo