circuitikz/circuitikz #166: Invert input of Logic Gates

Galva101, 2019-03-14:
Hi
I have run into an issue when I wanted to invert the input of a logic gate, below is a circuitikz example to try it with, ideally I would like to specify that for example input 1 of Nand Gate 1 should be inverted, as in the scanned picture below, without having to draw an empty circle and having to align it manually. I was sent here by Rmano over on [Stackexchange](https://tex.stackexchange.com/questions/479365/how-to-invert-logic-gate-input-in-circuitikz), so I hope this is the right place to post it.

Thanks very much, I hope I was able to explain it well.

[image]
[image]

    \documentclass[a4paper,12pt]{article}

    \usepackage{tikz} %vectorgraphics
    \usetikzlibrary{shapes,arrows,shadows,
    decorations.pathreplacing,backgrounds,calc} 
    %graph/flowchart
    \usepackage[siunitx]{circuitikz}

    \begin{document}
    \begin{circuitikz} \draw

    (0,0) node[nand port] (nand1) {}
    (3,0) node[nand port] (nand2) {}
    (5,0) node[nand port] (nand3) {}

    (3,2) node[nand port] (nand4) {}
    (5,2) node[nand port] (nand5) {}

    (3,4) node[nand port] (nand6) {}
    (5,4) node[nand port] (nand7) {}

    (0,4) node[nand port] (nand8) {}

    (nand1.out) |- (nand2.in 2)
    (nand2.out) -| node[circ,midway]{} (nand3.in 1)
    (nand2.out) -| (nand3.in 2)

    (nand8.out) |- (nand6.in 1)
    (nand6.out) -| node[circ,midway]{} (nand7.in 2)
    (nand6.out) -| (nand7.in 1)

    (nand4.out) -| node[circ,midway]{} (nand5.in 2)
    (nand4.out) -| (nand5.in 1)

    (-2,1) node(B1)[anchor=east] {B}
    (-2,1) -| node[circ,midway]{} (nand1.in 1) 
    to[short,*-] (nand1.in 2)
    (-2,1) -| (nand4.in 2)
    (1,1) node[circ]{} |- (nand6.in 2)

    (-2,3) node[anchor=east] {A}
    (-2,3) -| node[circ,midway]{} (nand8.in 2) 
    to[short,*-] (nand8.in 1)
    (-2,3) -| (nand4.in 1)
    (0.5,3) node[circ]{} |- (nand2.in 1)

    (6,1.97) node[anchor=east] {O2}
    (6,0.0) node[anchor=east] {O3}
    (6,4) node[anchor=east] {O1}

    ;\end{circuitikz}
    
    \end{document}

Rmano, 2019-03-14:
I will look into it - I have an idea that should be not so difficult to implement.

closed 2019-03-15 by Rmano

jagjordi, 2019-04-07:
What about European gates?

Rmano, 2019-04-07:
Have you tried? Add 
```
\draw (B.out) -- ++(0.5,0) node[european and port, anchor=in 1](D){};
    \node at (D.bin 1) [ocirc, left]{};
```
and you have: 

[image]

jagjordi, 2019-04-07:
I am aware of this possibility, but this is not consistent with the inverted output symbols for european gates, like NAND or NOR. I was thinking of the inclined line.

Rmano, 2019-04-07:
Ah, ok; (my problem is that I never used the European symbols...). Something like this? 
```
\draw (B.out) -- ++(0.5,0) node[european and port, anchor=in 1](D){};
    \draw (D.bin 1) ++(0,0.1) -- ++(-0.1,-0.1);
```

[image]

jagjordi, 2019-04-07:
Yes. It would be nice to be able to do it in a similar way as in `circuits` library from TikZ

Rmano, 2019-04-07:
I know, but this would be a quite complex thing to do; the implementation is really different. You can still hide this in a macro, however, like 
```
\def\negatein#1#2 { \draw (#1.bin #2)  ++(0,0.1) -- ++(-0.1,-0.1);}
``` 
(not tested) and then using `\negatein{D}{1}` ...

PR #168 (2019-03-15): Added border anchors to the logical ports.
This is especially useful to attach "ocirc" to negate
individual inputs.

Fixes #166