circuitikz/circuitikz #173: Toward a new release

Rmano, 2019-03-26:
Hi, @mredaelli @sistlind @der-stefan:

I am thinking about doing a release soon. I still would like to 

- [ ] do the background filling thing and testing it
- [ ] add a titlepage to the manual with a nice drawing

and I would like to add something of the @fotesan  contribution, but I am not sure on how to do it preserving authorship (the ground symbols, for example, or the elliptic generators --- although I do not know where they are used). 

Then I think I'll name it 0.9.0 and then, as soon the release is out, bump the git version number to 0.9.1 (to avoid confusion).

Rmano, 2019-04-14:
...and I should stop adding things, otherwise, I'll never release...

sistlind, 2019-04-15:
I thinks its worth waiting for Texlive 2019 release, which planned release
date is end of april. There should not come up any problems, but you never
know :)

Rmano, 2019-04-15:
Hmmm, yes,  it's a pity not being able to hit it, but I do not think there is time now.
My idea is finishing the instrument things and then sort-of freeze it during a time. We can check all is OK with TeXlive 2019 and then upload to Ctan.

Rmano, 2019-04-25:
> I thinks it's worth waiting for Texlive 2019 release, which planned release date is end of april. There should not come up any problems, but you never know :)

Stefan, would be triggering a new build after the release of TeXLive 2019 sufficient? In my computers setup I won't have the upgrade... or can you test it?

sistlind, 2019-04-25:
I am not sure which version the build system is using.
I have no TL2019 installation, it will be released at april, 30th:
https://tug.org/texlive/

I am not sure if they have a new upgrade procedures, previously there was
always a fresh installation necessary to upgrade to a new release.

Regards Stefan

sistlind, 2019-05-01:
Hi Romano,
I updated my local installation and it looks good with TL2019.

btw: the manual looks really cool now with the colored nodes.

Regards Stefan

Rmano, 2019-05-04:
Nice to hear! 
I am on holiday now, but I'll start the release thing next week. And thanks for the nice words!

Rmano, 2019-05-07:
Ok, I think that the last step missing is to prepare the release. 

@sistlind  suggests the following procedure:

- [ ] set version number and date in circtuitiks.sty, circuitikz-t.sty, manual and changelog
- [ ] commit everything
- [ ] run "make ctan" this will produce a zipfile. check the contents of this file (especially the version & date of docs & package)
- [ ] go to page https://ctan.org/pkg/circuitikz?lang=de
- [ ] click on upload(at the top)
- [ ] upload the zipfile and fill the missing fields ( I copied the changelog to the announcement field)
- [ ] May add a comment that you are the/a new maintainer to the administrative notes

Now the repository is freezed; I'll wait for:

- [ ] As this is your first upload, most probably the maintainer will send the other maintainers an email to verify if you are allowed to upload the package
- [ ] If the package is ok it will take a few days to spawn over all servers

And after that:

- [ ] Push your commit to github
- [ ] Add a github release with same infos & zip file.
- [ ] Add at the the link to the circuitikz-git file of the new release to the circuitikz github pages site (edit https://github.com/circuitikz/circuitikz/blob/gh-pages/index.html manually )
- [ ] reinsert the git headline in the changelog file.

...and now development will continue.

(Copied here to be able to track it, I hope @sistlind  do not mind).

Rmano, 2019-05-10:
I have prepared (and tested on a clean user) the CTAN submission zipfile. I am attaching it here hoping that someone can doublecheck it... 

[circuitikz.zip](https://github.com/circuitikz/circuitikz/files/3165981/circuitikz.zip)

Rmano, 2019-05-10:
The CTAN file is uploaded.

Rmano, 2019-05-11:
The CTAN announcement arrived today `Sat May 11 20:20:04 CEST 2019`:

```
Romano Giannetti submitted an update to the

                circuitikz

package.

Version number: 0.9.0 2019-05-10
License type: lppl gpl

Summary description: Draw electrical networks with TikZ

Announcement text:
----------------------------------------------------------------------

Version 0.9.0 (2019-05-10)
    - Added Romano Giannetti as a contributor
    - Added a CONTRIBUTING file
    - Added options for solving the voltage direction problems.
    - Adjusted ground symbols to better match ISO standard, added new symbols
    - Added new sources (cute european versions, noise sources)
    - Added new types of amplifiers, and options to flip inputs and outputs
    - Added bidirectional diodes (diac) thanks to Andre Lucas Chinazzo
    - Added L,R,C sensors (with european, american and cute variants)
    - Added stacked labels (thanks to the original work by Claudio Fiandrino)
    - Make the position of voltage symbols adjustable
    - Make the position of arrows in FETs and  BJTs adjustable
    - Added chips (DIP, QFP) with a generic number of pins
    - Added special anchors for transformers (and fixed the wrong center anchor)
    - Changed the logical port implementation to multiple inputs (thanks to John Kormylo) with border anchors.
    - Added several symbols: bulb, new switches, new antennas, loudspeaker, microphone, coaxial connector, viscoelastic element
    - Make most components fillable
    - Added the oscilloscope component and several new instruments
    - Added viscoelastic element
    - Added a manual section on how to define new components
    - Fixed american voltage symbols and allow to customize them
    - Fixed placement of straightlabels in several cases
    - Fixed a bug about straightlabels (thanks to @fotesan)
    - Fixed labels spacing so that they are independent on scale factor
    - Fixed the position of text labels in amplifiers

----------------------------------------------------------------------

This package is located at 
   http://mirror.ctan.org/graphics/pgf/contrib/circuitikz

More information is at
   https://www.ctan.org/pkg/circuitikz

------------------------------------------------------------------------

   Thanks for the upload.

     For the CTAN Team
    Erik Braun

------------------------------------------------------------------------

We are supported by the TeX users groups.   
Please join a users group; see http://www.tug.org/usergroups.html .
```

Rmano, 2019-05-11:
The remaining point will be ticked when merging PR #199 --- I have slightly changed policy here; bumping the number to 0.9.1 will ensure to be able to differentiate always between released version and git one. 

This is almost needed because I would like to implement the mechanism described in https://www.latex-project.org/publications/2018-FMi-TUB-tb122mitt-version-rollback.pdf , which will simplify the backward-compatibility restrains.

Rmano, 2019-05-12:
Release done, setup for continuing development done. 
Will close this issue.

closed 2019-05-12 by Rmano