circuitikz/circuitikz #176: Request: Oscilloscope

nobrl, 2019-04-02:
If I am not mistaken there is no oscilloscope symbol in the current package. I did find a scope symbol on  [stackexchange  ](https://tex.stackexchange.com/questions/209090/how-can-i-include-an-oscilloscope-in-my-circuitikz-circuit-diagram). 
Is it possible to add it to the next version of the package?

Here is the code written by Sebastian Golat :

> 
> ```
> \makeatletter
> \def\TikzBipolePath#1#2{\pgf@circ@bipole@path{#1}{#2}}
> 
> \newlength{\ResUp} \newlength{\ResDown}
> \newlength{\ResLeft} \newlength{\ResRight}
> \ctikzset{bipoles/SCOPE/height/.initial=.60}
> \ctikzset{bipoles/SCOPE/width/.initial=.60}
> \pgfcircdeclarebipole{}
> {\ctikzvalof{bipoles/SCOPE/height}}
> {SCOPE}
> {\ctikzvalof{bipoles/SCOPE/height}}
> {\ctikzvalof{bipoles/SCOPE/width}}
> {
>     \pgfsetlinewidth{\pgfkeysvalueof{/tikz/circuitikz/bipoles/thickness}\pgfstartlinewidth}
>     \pgfextracty{\ResUp}{\northeast}    %ResUp make usable
>     \pgfextractx{\ResRight}{\northeast} %ResRight make usable
>     \pgfextractx{\ResLeft}{\southwest}  %ResLeft make usable
>     \pgfextracty{\ResDown}{\southwest}  %ResDown make usable
>     \def\pgfcircmathresult{\expandafter\pgf@circ@stripdecimals\pgf@circ@direction\pgf@nil}
>         \ifnum \pgfcircmathresult > 45 \ifnum \pgfcircmathresult < 135
>             \pgftransformrotate{270}
>         \fi\fi
>         \ifnum \pgfcircmathresult > 135 \ifnum \pgfcircmathresult < 225
>             \pgftransformrotate{180}
>         \fi\fi
>         \ifnum \pgfcircmathresult > 225 \ifnum \pgfcircmathresult < 315
>             \pgftransformrotate{90}
>         \fi\fi
>     \pgfpathmoveto{\pgfpoint{0.75\ResLeft}{0.25\ResDown}}
>     \pgfpathlineto{\pgfpoint{0.05\ResLeft}{0.25\ResUp}}
>     \pgfpathlineto{\pgfpoint{0.05\ResLeft}{0.25\ResDown}}
>     \pgfpathlineto{\pgfpoint{0.65\ResRight}{0.25\ResUp}}
>     \pgfpathlineto{\pgfpoint{0.65\ResRight}{0.25\ResDown}}
> 
>     \pgfpathellipse{\pgfpointorigin}{\pgfpoint{0}{\ResUp}}{\pgfpoint{\ResLeft}{0}}
>     \pgfusepath{draw}
> }
> \def\SCOPEpath#1{\TikzBipolePath{SCOPE}{#1}}
> \tikzset{SCOPE/.style = {\circuitikzbasekey, /tikz/to path=\SCOPEpath, l=#1}}
> \makeatother
> ```

Rmano, 2019-04-02:
I am a bit confused about that symbol; I can't really fathom what "insert an oscilloscope in series" can ever mean. Maybe could be better to add a kind of optional "waveform" symbol to the ammeter/voltmeter?

nobrl, 2019-04-02:
I can't fathom etheir what "insert an oscilloscope in series" can ever mean. But I found my self in a situation where I needed an oscilloscope symbol. The "waveform" symbol to the ammeter/voltmeter seems a good solution to me (exepte maybe for differentiel probes).
[image]

Rmano, 2019-04-02:
Ok, modifying the ammeter/voltmeter is quite easy. I'll look into it (if you don't beat me) after I am finished with the work on #175. 
The only fine thing is to "unrotate" the waveform symbol; the solution here works only for multiple of 90 degrees, but we can use the solution from https://tex.stackexchange.com/questions/473544/unrotate-part-of-a-pgfdeclareshape

Rmano, 2019-04-11:
What do you think of this: 
[image]

The graph (apart for the manual description boxes)  is obtained with 
```
\begin{circuitikz}
        \foreach \a in {0,45,...,350} {
            \draw (0,0) to[oscope] (\a:3);
        }
        \draw (4,1) to[oscope, fill=green!20!gray, name=O1] ++(3,0);
        \path (O1.right) node[ground, scale=0.5, below right=4pt]{};
        \draw (5,-1) node[oscopeshape, fill=yellow!20!orange](O2){};
        \draw (O2.in 2) to[short, *-] ++(0,-0.5) node[ground]{};
        \draw (O2.in 1) to[short, *-] ++(0,-0.5) -- ++(-1,0) node[currarrow, xscale=-1]{};
\end{circuitikz}
```

nobrl, 2019-04-11:
That's much more than what I was proposing. Thanks a lot.

PR #179 (2019-04-11): Add the oscilloscope component
Fixes #176 

And while ai it, fix another bug in the manual
closed 2019-04-11 by Rmano