circuitikz/circuitikz #190: connectors

tcassanelli, 2019-04-18:
I see that you have implemented BNC connectors, I would also need [SMA](https://en.wikipedia.org/wiki/SMA_connector) and[ N type](https://en.wikipedia.org/wiki/N_connector) connectors. Do you have suggestions on how to create this two types of connections?

Rmano, 2019-04-19:
I added a part in the manual explaining how to add a component if you like to give it a shot. I am not expert in RF electronics, and in the pages you linked there is no symbol associated; from a fast search in google images it seems that often SMA connectors use the same symbol of BNC, with SMA specified aside. Can you point to some standard symbol?

tcassanelli, 2019-04-21:
Yes, couldn't find anything so far. I will keep looking and if I find something I will make a new symbol.

closed 2019-04-21 by tcassanelli

Rmano, 2019-04-21:
I think that all coaxial component share the same symbol. Maybe the name of the symbol was a bit of a mistake and should have been called `coax`...